d-ronin/dRonin #217: Packagework1 Cleanup: Packaging should do implicit clean of package at beginning

mlyle, 2015-12-10:
Cleanup from #177

If you build multiple packages from one tree/revision, there's conflicts in the same paths being used.

Packaging should remove the directory to be used at the beginning of a package highlevel target (package_all_compress, package_installer).

PTDreamer, 2015-12-10:
Can you clarify about the issue. Packaging dir name contains sha so there shouldn't be any conflicts besides dirty branches.
The reason why I didn't clean dir on the original PR was so you can have different packages from different branches or revisions living on the same build directory allowing for behavior comparison. 
Ok now I see that the issue clearly states multiple packages from one revision, but still the above still applies, same sha, same code, shouldn't be any conflicts.

mlyle, 2015-12-10:
@PTDreamer there's at least some degree of conflicting commands-- some are my fault (e.g. zip extraction of flight firmware).  I can work on seeing if it's necessary.

At the same time it kind of makes sense to me that if you are packaging the same release, to "start over" from a previous one (if it was ctrl-c'd, etc)

mlyle, 2016-01-15:
I'm happy enough with how things work, given that Jenkins integration is robust and works across all plats.

closed 2016-01-15 by mlyle