d-ronin/dRonin #228: Zero/low throttle stabilization. a.k.a borisb 'air mode'

fujin, 2015-12-10:
Discuss and potentially implement stabilization at low/zero throttle to aid with hang-time manoeuvres.

References https://github.com/d-ronin/dRonin/issues/182

dustin, 2015-12-10:
Cons:  Makes landings harder/less safe.
Pros:  Don't need it.

mlyle, 2015-12-10:
I think we can do something better without the downsides, as mentioned in the other issue.

dustin, 2015-12-10:
@mlyle Can you summarize here?  I'm a little dubious, but people seem to want it (I mean, it sounds OK when I'm not landing, at least).

I'm also a little curious as to how the history has unfolded.  People built flight controllers and wrote stabilization loops.  They added specific code to disable stabilization at zero throttle.  Cleanflight added a feature to override that and continue to allow stabilization at zero throttle.  Betaflight forked cleanflight and created that feature again.

It might be worth having a short writeup that describes why each of those decisions were made (i.e. what went wrong that required that change) along with whatever is done here and how it shows that we're not just repeating past mistakes.

mlyle, 2015-12-11:
@dustin : So, basically, we need logic to do smart things when commands clip.

If you have a throttle setting of 5%, and there's a pitch command of magnitude 10%, and say a yaw command of magnitude 1%, here's what happens:

The firmware spins the front left rotor at 14%.  It spins the front right rotor at 16%.

It wants to spin the rear left rotor at -4% and the rear right motor at -6%.  This doesn't happen and they spin at minimum speed.   So when there's clipping (either against full throttle or no throttle) the desired command doesn't get achieved.

In this case, the thrust delivered is (.14 + .16) / 4 = 7.5%.  This is more than the user asked for, and a reason when there's big oscillations or big commands multirotors tend to climb (along with actuator nonlinearity and the fact actuators may speed up faster than they slow down).

The pitch command is (.14+.16)/4 = 7.5% instead of the desired 10%.

The yaw command is (.02)/4 = 0.5% or half of what the user asked for.

The roll command is also 0.5% despite the fact the user asked for nothing.

So this introduces all kinds of weird cross-coupling between axes (the big command crowding out little commands) and can deliver more power than the user wants, as long as the user requests .001 or more throttle.

Basically my proposal is to partially fix this by scaling commands more intelligently and by limiting the amount of power the controller is willing to add for stabilization to 3% over the user's command.  Then, in turn, we can enable stabilization at no throttle (perhaps only for a very limited time, where that 3% decays down to 0% after 3 seconds at zero throttle).

mlyle, 2015-12-11:
(Also integral term management is necessary).

tracernz, 2015-12-11:
So my understanding is that ghettomixer already took care of this?

mlyle, 2015-12-11:
Ghettomixer doesn't provide a power limit for increases, and isn't factored right.  But it should be part of a major actuator update post-release

mluessi, 2015-12-11:
@mlyle sounds like a good plan :)

mluessi, 2015-12-11:
I tagged this post-renatus, but I wounder if for the short term something similar to air mode can be achieved by just setting the minimum throttle to 0.05 .. 0.1 or so when armed.

Robot-John, 2016-02-10:
I want this. Really bad. What can I do to help?

mlyle, 2016-02-10:
@Robot-John Cool! :)  I'm away for a couple of weeks but I'll be working on it when I get back.  We've already laid the groundwork with some of the clipping management stuff we've done.

One thing you could do is try out recent next, like http://jar.lyle.org:8080/job/dronin/152/  (click on the OS at the bottom).  More mileage on it will make us more confident to innovate.  And we have a big IRC community linked to from the website if you want to stop by and see what's going on.

mlyle, 2016-02-25:
@Robot-John --- if you're prepared to fly experimental code, #676 has an implementation of this.

Robot-John, 2016-02-25:
@mlyle - Awesome! Too bad a ton of snow just arrived. I'll get it installed and try some prop free bench testing.

PR #676 (2016-02-29): Low power stabilization
Fixes #228

* [x] Basic infrastructure for low power stabilization
* [x] Corrections to timer math for safety -- this was actually OK (proper windowed unsigned int subtraction)
* [x] Negative-side clip management implementation -- right now control authority is limited in that case
* ~~Need to give thought to integrator wind-up considerations in clipping scenarios~~ [issueopened]
* ~~Consider adding default actuator curve fit value = 0.85 or 0.9~~ [issueopened]
* ~~Consider adding actuator curve fit to UI~~ [issueopened]
* ~~Consider adding lowpower stab settings to UI~ [... cut]
closed 2016-02-29 by dustin