d-ronin/dRonin #236: GCS: select flash file window opens behind main window

jhitesma, 2015-12-11:
rev e58985a0 on Ubuntu 15.10

When I'm reflashing a board if I have GCS full screen and hit "Open File" the new dialog opens behind the maximized main window making it kind of a pain to use.

tracernz, 2015-12-11:
Interesting, this issue doesn't occur on 14.04.

jhitesma, 2015-12-11:
Upon further testing it also happens if I'm not full screen, just the main window is big enough to block the file open dialog.

However...it doesn't happen the first time.  

First time I open the "Open File" dialog it open on top like it should.  But if I close it or select a file and the hit "Open File" again then it opens behind the main window.

tracernz, 2015-12-11:
Okay, that occurs on 14.04 too.

VArcht, 2015-12-17:
Possibly related open issue here https://bugreports.qt.io/browse/QTBUG-46626

tracernz, 2016-03-20:
I tried this and still same result (dialog is behind main window on second and subsequent uses):
```cpp
/**
 *Opens an open file dialog pointing to the file which corresponds to the connected board if possible
 */
QString UploaderGadgetWidget::LoadFirmwareFileDialog(const QString boardName)
{    
    QString fwDirectoryStr = getFirmwarePathForBoard(boardName.toLower());

    QFileDialog dialog(this);
    dialog.setFileMode(QFileDialog::ExistingFile);
    dialog.setWindowTitle(tr("Select firmware file"));
    dialog.setDirectory(fwDirectoryStr);
    dialog.setNameFilter(tr("Firmware (fw_*.tlfw);;"
                            "Bootloader Update (bu_*.tlfw);;"
                            "All (*.tlfw *.opfw *.bin)"));

    QEventLoop loop(this);
    connect(&dialog, SIGNAL(finished(int)), &loop, SLOT(quit()));
    dialog.show();
    dialog.raise();
    dialog.activateWindow();
    loop.exec();

    return (dialog.result() == QDialog::Accepted) ? dialog.selectedFiles().at(0) : QString();
}
```

mlyle, 2016-03-24:
Won't be further improving this before samsara; deferring.

(Less important with auto-upgrade).

mlyle, 2016-10-07:
Same rationale used to defer again.

tracernz, 2017-06-20:
I can't find any reports of this from other Qt users. That + my issues with Qt OpenGL stuff within GCS  main window/uavgadget vs in standalone apps makes me quite suspicious of our main window + uav gadget stuff...

jhitesma, 2017-06-20:
I'm not even sure if I can repo this anymore...the linux notebook I was using it on where this was happening is 32bit so I abandoned it awhile back.  I haven't seen it happen on MacOS or Windows - and don't have a newer system running Linux to test on right now.

tracernz, 2017-06-21:
I've seen it recently but never found a reliable repro.

mlyle, 2018-03-18:
I haven't seen this, but here's one you can consistently repro:  if you have setup wizard open, and plug in a FC that needs upgrade or wants to spawn setup wizard, it opens in front of the main window but behind setup wizard, modal, and setup wizard freezes (and must be moved out of the way)

tracernz, 2018-03-18:
Can you open an issue for that? It's very different to this.

mlyle, 2018-03-18:
I think it's #1285 actually ;)   We have a lot of modal dialogs with no clear parent; we have a lot of things that can spawn modal dialogs at any time.

jhitesma, 2018-03-18:
Confirmed I still can't repo this on mac and windows.  But still can't confirm or deny on linux because I still haven't setup a linux machine with a GUI to test on :(  If no one else has noticed and you guys can't repo I'm willing to assume it's no longer an issue.  ( I kind of wonder if it went away when we dropped support for 32bit since that was when I quit using my linux notebooks since they were all old 32 bit machines...)

PR #2174 (2018-04-23): gcs: qfiledialog: ensure there's always a parent
fixes remaining instances of #236

fixes #236
closed 2018-04-23 by tracernz