d-ronin/dRonin #249: GCS: firmware tab interface usability

jhitesma, 2015-12-12:
The buttons on this tab kind of need a full rethink.

The way they appear and disappear isn't very good UI design.

And it's possible to do unintended things.  Like if you hit Halt then "Reset" goes away but "Open File" is still available and active.  Clicking "Open File" opens the open file dialog...but the halt process gets lost.  The board appears to halt, but the GCS never notices and you're left in a funky state.  

At a minimum we should disable "Open File" while a halt is in progress.  But rethinking the entire tab is probably a good long term goal.

tracernz, 2015-12-12:
I've been meaning to do something about this. I hate how the buttons appear and disappear too.

mlyle, 2016-03-01:
Mostly fixed by #677

closed 2016-03-01 by mlyle