d-ronin/dRonin #260: GCS:Configuration: Scrollbar obscures tabs

tracernz, 2015-12-13:
This was reported in TauLabs/TauLabs#1146 and is relevant on Linux as well (OS X is okay due to the native scrollbar style used). The issue is becoming worse as more and more tabs get added. Should be easy enough to fix by applying a CSS style.
[image]

tracernz, 2015-12-14:
For reference, OS X:
[image]

mlyle, 2015-12-14:
The styling on OSX is bad too because if it lines up wrong you may not see there's anything to scroll too (since the scrollbar auto-hides)

tracernz, 2015-12-14:
If we made it have the OSX style and remain visible all the time it could be good.

PR #378 (2016-01-06): gcs: attempt to make config tab icons work better
Fixes #371 . Fixes #260 .

[[image]](https://reviewable.io/reviews/d-ronin/dronin/378)
PR #441 (2016-01-15): GCS:Config: Restyle widget scrollbar similar to OSX
Should stop the scrollbar hiding the icons and labels when it is present. Tested on Linux, needs tested on Windows and OS X (OS might ignore it there I think).

Fixes #260 more.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/441)

Needs:
* [x] Linux test
* [x] Windows test
* [x] OS X test
closed 2016-01-06 by mluessi