d-ronin/dRonin #283: packaging: Confirm GCS finds firmware on each platform

mlyle, 2015-12-17:
Probably requires special attention on OSX

dustin, 2015-12-17:
Ideally, the GCS doesn't ship with firmware and just grabs what's needed remotely.

It's a huge annoyance when I want to update, e.g. a Brain and I have to redownload the entire GCS just to get to the firmware that's included with it.

At the very least, package them separately so I can download just a firmware image when I need it.

mlyle, 2015-12-17:
Right now firmware and GCS need to be versioned together because of various dependencies.  You can't (reliably) configure a board from a GCS of a different version.

So right now (and this isn't going to change soon) you need to download both, and the firmware doesn't add much size to the GCS installer image (such that it would be an advantage to only download one).

mlyle, 2015-12-23:
It looks like the wizard / resource usage in uploader is wrong since we removed that stuff.   Though it often didn't seem to work for me before anyways.

We should also consider putting the firmware inside the mac application bundle.  (With "firmware" being a symlink or some such, even though that's naughty).  Then uploader can find firmware on mac.

tracernz, 2016-02-15:
I can get this working fine on Linux in both the setup wizard and uploader file dialog, but not on OS X. It seems like OS X just totally ignores the path you provide for the file dialog.

tracernz, 2016-02-15:
BTW, my hax: https://github.com/tracernz/dronin/tree/firmwarez

mlyle, 2016-02-15:
I have no Mac with me.  But it may expect a macos style path-- e.g. colon delimited.  I know qt has path and path separator related functions.

tracernz, 2016-03-19:
This one is taken care of in upgrader work.

tracernz, 2016-03-20:
Should be fixed by #801 (or previous upgrader work).

PR #801 (2016-03-24): GCS: Upgrade assistant
Provides an upgrade assistant that retrieves settings from the quad and sends them to the autotown cloud service to be adapted for the current release, and then upgrades the quad to the latest rev.

Fixes #3 
Fixes #83 
Fixes #283 

---
This change is [[image]](https://reviewable.io/reviews/d-ronin/dronin/801)
closed 2016-03-24 by mluessi