d-ronin/dRonin #284: Renatus Technology Preview Release 2 Issue

mlyle, 2015-12-17:
This is a roll-up of stuff needed for R-TPR2

* [ ] #213 License.txt shipping and License.Txt cleanup
* [ ] CONFIRM packaging: MacOSX/DMG
* [ ] CONFIRM packaging: Windows/Installer
* [ ] CONFIRM packaging: Windows/zip
* [ ] CONFIRM packaging: Linux/tar
* [ ] CONFIRM packaging: Linux/deb
* [ ] NiceToHave: #237 Config widget behind splash screen fix
* [ ] NiceToHave: #157 / #279  RTH sequencer merged so we can hopefully get test feedback from @pug398 and others
* [ ] NiceToHave: #283 GCS finding firmware on all plats (or at least knowing the weaknesses)

dustin, 2015-12-17:
I'd argue for #75 -- it should be trivial to get a small request in similar to what comes in from autotune results.  Perhaps this can be issued when a board is connected.

dustin, 2015-12-17:
Also, #287 seems easy and useful.

mlyle, 2015-12-17:
@dustin Are you familiar with the GCS side of those interchanges at all?  We don't have @tracernz and I'm fairly slow on the GCS side of things.  I really want to get a preview release cut in the next day or two (and expect I'll be rather busy with packaging) -- because we'll get much better feedback with where we are now than the last hastily-cut preview release.

dustin, 2015-12-17:
I'll take a look.  It's stuff I want, there are examples, and I can build.

mlyle, 2015-12-18:
Packaging confirmations ticked off in conjunction with #289

mlyle, 2015-12-18:
User spastic from IRC has used the Windows prototype installer.   Encountered issue #295, but otherwise seems to be doing well.

5PAZT1C, 2015-12-18:
migrated from tau labs to this release. so far so good, havent flown it yet. Excited about trying autotune on my cc3d

closed 2015-12-19 by mlyle