d-ronin/dRonin #288: POItracking to brushless gimbal via MAVLink

Ga1j1n, 2015-12-17:
Existing code (Potentially removed) is aimed at direct servo control, Propose enhanced POITracking control via MAVLink or STorM32 protocol directly to STorM32 brushless gimbal.

Project should re-enable and modify impressive position tracking / following performance and the STorM32 could fill the role of the SparkyBGC with a readily available, low cost and opensource project

https://github.com/TauLabs/TauLabs/wiki/User-Guide:-POI-tracking

https://vimeo.com/127901407

http://www.olliw.eu/storm32bgc-wiki/Main_Page

jihlein, 2015-12-17:
Would love to see this for STorM32 gimbal controller, been thinking about it myself.  I have a STorM32, not sure where to start with code yet, but happy to help test and any other way I can.

mlyle, 2015-12-17:
It's probably not too hard to get the POI information shoved into Mavlink.

PR #723 (2016-03-05): Flight: Storm32bgc
Simple control of Storm32bgc.  Enables sending cameraDesired.Declination and  cameraDesired.Bearing to gimbal controller over serial link.

A UAVObject for the controller's ACK response was added to make it easy to verify proper communication with Storm32bgc is occurring.  A value of 1 indicates successful ACK, all other values indicate an error of some sort occurred.

Implemented for F4 and F3 targets only.

Tested on AQ32, Quanton, and Sparky targets.  Could only test  cameraDesired.Declination since I only have a 2 axis gimbal (pitch/roll) setup at the moment.  No reason  [... cut]
closed 2016-03-05 by mlyle