d-ronin/dRonin #291: Win 10 GCS Next crashes with "Netbook" selection

ufoDziner, 2015-12-17:
GCS crashed when selecting "Netbook" from the layout? screen after a fresh "next" compile on Windows 10.  The "Developer" option worked.

mlyle, 2015-12-19:
Seems to work on Mac.  Will try on Win10 next.

mlyle, 2015-12-20:
Can you repro this?  I can't reproduce.  Just tried netbook layout on Win10.

ufoDziner, 2015-12-22:
I will attempt to when I get home.  How do I clear out the default that it now uses?

mlyle, 2015-12-22:
There is a menu item to import/export gcs settings.  It also lets you reset
the config.

The windows installer also adds a shortcut to clean the config but that was
broken until a couple days ago.

ufoDziner, 2015-12-24:
Reproduced after resetting the config.  Screenshot attached.
[image]

mlyle, 2015-12-24:
@ufoDziner Can you try it from the installer package to see if it works on your machine that way?

ufoDziner, 2015-12-24:
I can, but need some direction as to how to compile it for a package, and/or find it in the build to install.  Sorry for the hassle.

mlyle, 2015-12-24:
Preview release packages are here:

https://github.com/d-ronin/dRonin/releases/tag/preview-20151218

No hassle-- all bug reports are valuable and your help nailing this down is useful.

If it works in the installer, it's primarily a pain for devs and we'll defer this issue out of Renatus, though, I think.

ufoDziner, 2015-12-24:
That release installed package works just fine with the netbook configuration.

mlyle, 2015-12-24:
OK.  So there's probably a problem with some of the paths and how they work in the build directory configuration.  Deferring from Renatus.

mlyle, 2016-01-01:
Duplicate #352

closed 2016-01-01 by mlyle