d-ronin/dRonin #295: Need prominent OpenGL / Mesa on windows errata

mlyle, 2015-12-18:


mluessi, 2015-12-22:
Can you give more details? It isn't clear to me what this means from the title alone.

mlyle, 2015-12-22:
People still are tripping over the thing where with old Intel and other chipsets that don't ship an OpenGL implementation that the GCS doesn't work.  We need a prominent errata and instructions on how to deal with it.

mlyle, 2015-12-31:
@PTDreamer -- do you have a fix for this or did you just encounter it?

PTDreamer, 2015-12-31:
Hi have the dump where it throws the exception so I can try and have behave
more graciously. Also I had a PR once to detect the need for the sw mesa
and download it and install it if needed.
Also there seem to be two different issues, one that causes some widgets
not to be rendered and another which causes a segfault.

mlyle, 2016-03-24:
Won't be further improving this before samsara; deferring.

PR #1145 (2016-07-26): Build: Windows build improvements, VS2015, Pick up QT 5.6.1
Fixes #295 
Fixes #373 
Connects to #290
Supersedes #1142
closed 2016-07-26 by tracernz