d-ronin/dRonin #296: "0 module is in warning state"

mlyle, 2015-12-18:
Need to run this down / make it helpful.  @5PAZT1C

5PAZT1C, 2015-12-18:
I cant find any indication of what this warning is in the GCS. everyhing seems to be working fine, maybe this is just a bug? is it specific to cc3d?

jhitesma, 2015-12-18:
I've seen it on Naze32 and Sparky so it's not just cc3d.  Probably on everything.

pug398, 2015-12-19:
This is inherited and on many platforms. Has not affected flight but has been previously mentioned. It is good you would like to try and get it fixed. The problem is not in TL 6e3d but is in TL 630C. I will try to narrow the gap...

pug398, 2015-12-19:
The closest I can get at the moment is 20151007 b0fc is good and 20151014 555f introduces "module 0" error. A few changes in about a weeks time for you to sort through.

mlyle, 2015-12-19:
Yah, unfortunately there's nothing interesting in those 5 PRs.  I think I'd seen it before.  Will add an assert to the UI to see if I can figure out why.

SystemsGuy, 2015-12-19:
Confirmed on Sparky1 on pre-release. 
[image]

mlyle, 2015-12-19:
OK, so this one is figured out / there's a PR open.   Basically, there's a bunch of other fields in systemsalarm that if they weren't 0 or 1 would cause this message because we scanned the structure wrong.  There was a change in that revision range that made it so we almost always have a field with value not 0 or 1.

PR #310 (2015-12-20): sysalarms: only look at correct uavo field
Fixes #296 

Previously we'd iterate over all the stuff in SystemAlarms instead of just the stuff that represents alarms.

This would trigger on various configuration errors and other things "0 module in warning state".  With Kenn's change to report reset reason, it would always fire on normal powerup.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/310)
commit 100d9361 (2015-12-20): sysalarms: only look at correct uavo field

Fixes #296
closed 2015-12-20 by mluessi