d-ronin/dRonin #298: Support the Naze32 rev6

hernick-qc, 2015-12-18:
dRonin should support the new Naze32 revision 6 board, which is now the only official Naze32 you can get from many official Abusemark resellers. It doesn't seem like there's a lot of work required to get this working.

The main difference is the use of the MPU-6500 instead of the MPU-6050, and this new chip is already supported by the BrainFPV MPU-9250 i2c driver.

Tracernz has an example hack to get the Naze32 rev6 working at https://github.com/tracernz/TauLabs/tree/naze-rev6-hax

It also uses the BMP280 barometer, for which jihlein has written a driver that can be found on his TauLabs fork at https://github.com/jihlein/TauLabs/tree/myNext/flight/PiOS  - Common/pios_bmp280.c inc/pios_bmp280.h and inc/pios_bmp280_priv.h

hernick-qc, 2015-12-18:
I'd like to try to get this working over the holidays, and I'd like some guidance as to the best approach.

Should all drivers (MPU-6050, MPU-6500, BMP280, etc..) be included in all naze32 firmware builds, with onboard autodetection of the proper driver to use?

Or should this be a build-time setting (choosing between <=rev5 or rev6), with only the required drivers being included the firmware?

CptanPanic, 2016-02-22:
Any new updates for this?

tracernz, 2016-02-22:
Waiting on me to have time to finish driver still.

mluessi, 2016-02-22:
@tracernz which driver? I have a BMP280 one that I got from @jihlein, maybe he can make a PR for it?

tracernz, 2016-02-22:
Common mpu driver.

jihlein, 2016-02-22:
@mluessi I can make a PR for the BMP280 driver, but will most likely need to wait until the weekend, going business traveling again....

tracernz, 2016-02-23:
We don't currently support barometer on any version of the naze so no rush on that.

tracernz, 2016-03-20:
Fixed by #757.

PR #757 (2016-03-30): One MPU to rule Naze only, for now
Rebased to next (ouch, took a lot of work). Still needs lots of work but putting this up so it's visible. Mag support not yet working, if you can spot anything obvious for 9250, please let me know.

Works on Naze rev 5 & 6, 4 untested.

edit by mpl: fixes #298 

[[image]](https://reviewable.io/reviews/d-ronin/dronin/757)
closed 2016-03-30 by mlyle