d-ronin/dRonin #304: Report of wrong (TL) shortcut on Windows

mlyle, 2015-12-19:
from @pug398

pug398, 2015-12-19:
With a reboot this morning the TL desktop icon (with dRonin text) changed to dR. It was the first reboot after running dRonin PR2 installer.

mlyle, 2015-12-19:
Interesting.  So some remnants of TL stuck around.

mlyle, 2015-12-20:
I played a little with reproducing this and wasn't able to get it consistently.. I think we'll ship with this.  It seems to do the wrong thing (until reboot) if you're installing over Tau Labs.

VArcht, 2015-12-21:
I saw this when I was working on the branding, I think the OS is caching the icon? I googled some and didn't see a clear solution. Perhaps changing the icon name would get around it?

mlyle, 2015-12-21:
But we changed the name :/

VArcht, 2015-12-21:
hmm, right, does it get an alias somewhere?

mlyle, 2016-02-23:
Doesn't seem to happen anymore.  I think it's a short-lifetime thing after upgrade.

closed 2016-02-23 by mlyle