d-ronin/dRonin #308: GCS : Dead Hyperlink

SystemsGuy, 2015-12-19:
ESC Signal selection screen in GCS has "more information" link which links back to d-ronin github that then points to the deaded OpenPilot wiki
[image]

mlyle, 2015-12-19:
@hernick-qc Would you be willing to look at this?  It'd be a big help :)

hernick-qc, 2015-12-19:
I'll take it.

Actual issue is that there are still many links to the old OpenPilot wiki inside the dRonin wiki. I will update the OnlineHelp: pages' content to include useful information rather than dead links.

tracernz, 2015-12-19:
As @hernick-qc says, there are quite a number of Online Help pages that point to the now dead OP wiki.

mlyle, 2015-12-19:
@hernick-qc -- feel free to create subissues and maybe some of us can take some of the load.

(I can't really look at doc myself much though until closer to release date)

tracernz, 2016-01-14:
We should rename this page to OnlineHelp: ESC Refresh Rate or similar before our first release, doing it after will break old release help links. I removed the OP link and added a link to our Oneshot125 page for now. https://github.com/d-ronin/dRonin/wiki/OnlineHelp:-TurboPWM-ESC's

mlyle, 2016-01-18:
Renaming and deferring based on a lot of stuff done already.

joeynuggetz, 2016-03-18:
Also broken.......

Help button on "Output" page in GCS links to http://wiki.openpilot.org/display/Doc/Output+Configuration which is non-existent.

Broken link on the main help page for "Input" screen in GCS. Link to "Input Configuration" points to http://wiki.openpilot.org/display/Doc/Input+Configuration which is non-existent.

jpbarraca, 2016-03-22:
Should we/can we salvage some information from the old openpilot wiki to speed up these issues?

joeynuggetz, 2016-03-26:
All links that appear on bottom of the landing page after clicking on ? on the Vehicle page all point to old OpenPilot URLs:

[image]

mlyle, 2016-04-01:
Removed the last vestiges of openpilot from the wiki and improved documentation some.

closed 2016-04-01 by mlyle