d-ronin/dRonin #315: Tricopter setup incorrect after power cycle(s).

jihlein, 2015-12-19:
Setup AQ32 for Tricopter from clean eeprom.  PPM rx, batt monitor, FrSky sensor hub, openlog, gps, external mag.

On Vehicle tab, Vehicle type multirotor, Frame type Tricopter Y.  Motor output channels 1,2, and 3 set to Channel2, Channel1, Channel3, respectively.  Tricopter Yaw Motor channel set to Channel5.

On Output tab channel banks 1, 2,and 3 set to SyncPWM, SyncPWM, 330 Hz respectively.  Resolution 12MHz for all channels.  Channel min/neutral/max set at 120/131/250 for motor channels, 2000/1500/1000 for servo.

Flew basic stability checks, including autotune.  Flight and autotune went well.  Applied the autotune gains, and flew them.  After flight (and a few power cycles) while creating autotune report, noticed that the GCS reported my configuration was quadx.  Looking at the Vehicle tab, Quadx is now the selected frame type, the 3rd motor channel and tricopter servo channels are set to none.  Output tab is as originally configured, and motors and servo still respond correctly for the tricopter configuration.

Inspection of SystemSettings.AirframeType showa QuadX config.

After power cycling again, the same erroneous configuration is reported, but the vehicle still behaves as a tricopter, motors and servo working as configured.

mlyle, 2015-12-19:
Initially triaging into renatus, but probably punting out.

jihlein, 2015-12-21:
Can't replicate this issue any longer after rebasing to current next and rebuilding.  The legacy TauLabs issues are still there.  I've managed to fix the extra motor definition, but have not made any progress on the missing servo channel display.  Working on getting somewhat familiar with the GCS code.

I'll leave this issue open a little longer to see if the original problem reappears while testing.

jihlein, 2016-09-02:
Looking back into this, it appears to be caused by the fact that out of the SystemSettings->AirframeCategorySpecificConfiguration UAVO, only the 1st 32 bits out of the 128 bits of data are actually getting saved.  The tricopter servo configuration is stored in bits 33-36.  I've also noticed that in order for SystemSettings->AirframeType to be reliably saved, the save button on the GCS Vehicle setup page has to be selected multiple times.  I've tried saving the objects via the UAV browser window with no change in behavior.  Inspecting the code, it's not obvious to me where the problem lies at the moment.

mlyle, 2017-01-30:
Same as (later, but more information) #1220

closed 2017-01-30 by mlyle