d-ronin/dRonin #323: Large amounts of jitter on Sparky2 OneShot outputs

droidicus, 2015-12-20:
Attached are a few screenshots from an oscope, showing large variations in pulse-to-pulse spacing. The frequency of the oneshot pulses themselves is consistent, but the offset of some of the pulses seem to be variable in time.

[image]
[image]
[image]
[image]
[image]

droidicus, 2015-12-20:
These screenshots were taken on a Sparky2 running PR2 firmware, quad was idle.

mlyle, 2015-12-20:
This doesn't look too too bad-- individual delayed pulses.

mlyle, 2015-12-20:
In Renatus at least for now.

droidicus, 2015-12-20:
This is also causing pulses to merge together into a single, larger, pulse as well. This pulse is longer than the maximum oneshot pulse time by quite a bit (~404 us).

[image]

droidicus, 2015-12-20:
It appears that all of the channels are showing the same behavior in a synchronous manor. Here is channel 1 and 4:
[image]

AlessioMorale, 2015-12-21:
This is actually the same issue i have observed in our (LP) firmware. I've tracked down the problem (using several performance counters to measure individual chain delays) being an inconsistent latency from gyro interrupt to the actuator module update (although I've added a pulse skip logic to prevent the "pulse merge" issue, needed by Sync-PWM implementation). I guess (i'm working on that direction) that the best and easier solution is some kind of "fast lane" approach from sensor input -> stabilization inner loop -> actuator.

mlyle, 2015-12-21:
It certainly could make sense to more tightly integrate sensors and stabilization -- like we have on the F1 targets.

We used to have a pretty bad issue where sparky2 radio reinit was causing jitter, but @peabody124 has mostly fixed that.  The causes are probably slightly different due to drift between Tau/dR and LP, but probably this either relates to us doing something expensive with interrupts masked or a high priority task.  In dRonin right now, ManualControl, Stabilization, Actuator, and individual PiOS sensor tasks   are HIGHEST priority.  But sensors and attitude are HIGH.  So there's a lot of opportunities for something to run too long and create jitter.

droidicus, 2015-12-21:
looks like this is the OP/LP commit that implements the pulse skip for them, not sure if this is useful for us with the diverged codebase and all: https://bitbucket.org/librepilot/librepilot/commits/752f3447d2489cf6c0c4508bd72a970b0805257f

sim-, 2015-12-22:
Note that it sure would be nice if there were some guard time for minimum low time or else problems like this may cause pulses to "crash" together and cause signalling issues, as seen on other platforms: http://0x.ca/sim/tmp/cleanflight_oneshot_mind_the_gap.png . Even if not an electrical issue, many Atmel ESCs can't measure less than a microsecond or two, because the edge to be looked for next must be flipped in the receiving interrupt (no hardware PWM RX).

droidicus, 2015-12-23:
@sim, agree. The OP/LP has a 50 us minimum low period in their pulse-skip algorithm, that seems like a good value. Less then this wouldn't really be helpful anyway, even if the ESC was able to reliably read it.

tracernz, 2015-12-23:
Yep, we need something like that. 50 us or half of the minimum pulse-width would do.

droidicus, 2015-12-23:
@tracernz I like the idea of half the minimum pulse width even more, that way it will scale well if we go to oneshot42 or whatever else.

tracernz, 2015-12-23:
It would be a little limiting for Oneshot1000, `1 / (2000e-6 + 500e-6) = 400 Hz` would be okay though I guess.

mlyle, 2015-12-23:
Opened #337 for the more severe issue (pulse merge).  Deferring this one out of Renatus.

droidicus, 2016-02-06:
Fixed in #342

mlyle, 2016-02-06:
#342 fixed #337 (pulse merge) ; a need for a jitter study and RTOS work remains, particularly during logging

mlyle, 2016-09-09:
I scoped sparky2 .  occasional 150-200us jitters during logging.  I don't care that much. :P

closed 2016-09-09 by mlyle