d-ronin/dRonin #324: Map gadget should point to bing

mlyle, 2015-12-20:
Bing actually works well.  Google does not.

mluessi, 2015-12-22:
Can you elaborate? Google is working fine for me.

tracernz, 2015-12-22:
We are actually using the TL API key ATM and from what I understand the way the API is (ab)used is not in accordance with the TOS.

mluessi, 2015-12-22:
@tracernz yes, this is something that needs to be fixed.

mlyle, 2015-12-22:
@mluessi At some levels of zoom the OP map gadget works.  At full zoom out or some inbetween levels it doesn't work for me (just shows the default "background").  The bing settings work across all levels and load more quickly.  Does it work at full zoom-out for you with Google?

mluessi, 2015-12-22:
Google seems to be working for me. However, @tracernz brought up a good point. We shouldn't be using gmap if it is against their TOS. Even if we use it, we would need to update the API keys, as we currently don't have control over them. We should check the Bing TOS to make sure we are in the clear.

mlyle, 2015-12-22:
Google doesn't issue new maps api keys, I believe, too.

No one lets you embed maps free-- except open street map.  So technically
we need to pull everything, which we should probably do... But maybe ship
our first release close to as is before we  deeply audit things

PTDreamer, 2015-12-22:
The map graphics don't use/need an API key. It is only used for geocoding which is used even when Bing is set as the map provider.
Google maps are working fine for me,  I would try to clear the cache.

tgreer, 2016-01-07:
I've never had a reliable Map in GCS. it's always screwed up as soon as I've zoomed out.

PTDreamer, 2016-01-07:
Can you make a video of the issue?

mlyle, 2016-09-24:
Meh, it's been reliable for me recently.

closed 2016-09-24 by mlyle

PTDreamer, 2016-09-26:
Fyi api key is only used for geocoding (i.e. string to location) not for
map tile fetching

mlyle, 2016-09-26:
@ptdreamer -- Hey!  Glad you're still around.

Yes.  I think the main problem I was encountering is it was possible to try and fetch illegal tiles on the furthest zoom level out if your resolution was high enough and it made it do weird things.  We've corrected that (if the window is too big we force people to zoom level 2).