d-ronin/dRonin #337: Need minimum deadtime on syncpwm output

mlyle, 2015-12-23:
re: issue #323 

My proposed fix is, around here:

```
        if (output_channel_mode[servo] == SYNC_PWM_TRUE) {
                TIM_Cmd(chan->timer, DISABLE);
        }
```

To ensure that the count value is past max+10us.  This implicitly assumes the max values for the things in the same bank are the same.

It's super unfortunate that we're not using hardware oneshot/single count functionality which I think exists on all of our targets.

mlyle, 2015-12-24:
How it would work with hardware oneshot:

A) don't disable the timer explicitly there
B) set the count period to be that max+10us or max*1.25 or whatever value.
C) only enable timer if already disabled.

tracernz, 2015-12-24:
Yeah, that's what I had in mind when discussing a deadtime in the other issue. :+1:

PR #342 (2016-01-04): pios_servo/oneshot: Ensure dead time.
Now max pulse length is passed into pios_servo, and we make sure we're
"past" that point by a defined margin before we'll disable the timer.

pios_servo should get factored common in the future and non-HPWM
variants of the code should get removed in that refactor.

This depends on all channels in a oneshot bank being defined with the
same "max" time to be fully protective.

Fixes #337

[[image]](https://reviewable.io/reviews/d-ronin/dronin/342)
closed 2016-01-04 by mluessi