d-ronin/dRonin #343: RFC: Indentation & PR-checks

mlyle, 2015-12-24:
So, there's been a lot of discussion that we should have PRs verify correct indentation automatically.

It's difficult when there is a lot of existing code that is not indented properly.

There's been discussion that we should do a "big reindent" of the code, but this will conflict with work in progress.

I suggest this for post-renatus:

1. we introduce a PR check that makes sure that on already-perfectly-indented flight files that no lines are improperly indented.  This will minimize pain.  (We need to also be sure to exempt upstream stuff)
2. we re-indent PiOS and a few other infrequently changing places right away
3. we re-indent much more of flight near the end of the release if things go well.

For now we can ignore ground.

Thoughts?

dustin, 2015-12-24:
I'm very much in favor of having tools do this.  Spoiled by gofmt.

PTDreamer, 2015-12-24:
Doing this kind of cleanups kills attribution (obvious attribution i.e. the output of git blame). 
Why not just run the check against changed lines? Sure it will end up with files having mixed styles but it will increasingly get better while keeping proper attribution.

mlyle, 2015-12-24:
For the most part `git blame -w` will keep the correct attribution.