d-ronin/dRonin #350: "Upstream" tracking: need a magical merge commit

mlyle, 2015-12-25:
After #348 and #349's fates are known, we should do a merge from TL `-s ours` to "ignore" everything that has drifted since in future rebases.

Those commits, right now, are:

```
pick 92ebf05 GCS/OSX: updates from comments
     Not really applicable
pick 893ef25 Ground-Updated license template
     Different project / separate changes
pick cdfe9cc Fix project credits file
     Ditto
pick de0413c Fix project credits file
    Ditto
pick 0a505b4 More granular alarm messages (subtypes).
    We have this independently / factored cleaner
pick 30f4286 FlyingF4: Use new ADC
    N/A removed target
pick e56e200 GCS: USBInfo remove unused fields
    We have a different solution
pick b32e4a1 VehicleConfigHelper: add default value
    We already have this
pick d382b81 UAVObjects Plugin: Update copyright
    N/A
pick b6db52e Main: Update copyrights
    N/A
pick ce31fc0 SysAlarms: Update copyright
    N/A
```

mlyle, 2015-12-25:
@dustin What you think here could be particularly valuable.

dustin, 2015-12-25:
I'd just try it and see what showed up.  It could be awful, or do very
little. The diff would tell.

mlyle, 2015-12-25:
The goal would be zero-diff but to get the commits into the rev history so future rebases or merges would ignore them.

dustin, 2015-12-25:
Does "-s ours" keep the tree, or just resolve merge conflicts that way?  I
don't know that I've actually tried this.

mlyle, 2015-12-25:
I'm pretty sure it'll do the right thing.  The main problem (but also benefit) is we'll have things in our git log that were never actually done here / ignored.

dustin, 2015-12-25:
Well, they'll just have been undone.

PR #366 (2016-01-04): tracking-taulabs: squash all the history
Simplify / speed up future rebases back and forth, by getting all the changes we're not taking into our history.

fixes #350 

@dustin Is this a good idea, or should we punt on the idea?

[[image]](https://reviewable.io/reviews/d-ronin/dronin/366)
closed 2016-01-04 by mluessi