d-ronin/dRonin #352: Win 10 GCS Next crashes on startup

ufoDziner, 2015-12-26:
Running windows 10 on an older Alienware laptop.  Built a current "next", but am experiencing crashes on startup.  After the build, I navigated to build->ground->gcs->bin and double click on drgcs.exe.  The following is my git information, and a screenshot of the message I get when it crashes.

ufo@Wes-M11x MINGW32 ~/Documents/GitHub/dRonin ((2b9582a...))
$ git describe
preview-20151218-70-g2b9582a

[image]

mlyle, 2015-12-31:
Can you try resetting your config?   The installer has a shortcut for this, or -r as a command line argument to drgcs works

ufoDziner, 2015-12-31:
I reset the config with -r per your request.  It will then load if I select "Developer", and crashes if "Netbook configuration" is selected.

mlyle, 2015-12-31:
OK, I see.  So it's the combination of "netbook configuration" and running from your own build tree.  I'll check it out.

PR #365 (2016-01-03): gcs: configurations: housekeeping
Adjust the stylesheets to have more common ground.  Fix #352 (I hope).  Needs test.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/365)
commit 7a113010 (2016-01-03): gcs: netbook: adjust qml on uploader pane

To be consistent with other targets.

Hoped that this fixes #352
closed 2016-01-03 by dustin