d-ronin/dRonin #361: PipXtreme / OPLink binding broken / no telemetry transmitted to GCS Preview 2

Ga1j1n, 2015-12-29:
After updating firmware / GCS to preview 2 I can no longer go through the bind wizard to set the OPLink as co-ordinator, the device seems to connect to the GCS but no data is passed through.

Preview 1 on a seperate machine with the new firmware on both units works fine, suspect a bug in the GCS

mlyle, 2015-12-29:
Just to confirm-- did you upgrade the oplink firmware too?

ekaszubski, 2015-12-29:
I ran the bind wizard yesterday with latest next (2b9582a919fc6d8abccd8daf877070b157262697) on sparky2 + OPLM and everything worked flawlessly.

Procedure I used was:
- enter rescue mode
- plug in OPLM to USB
- flash firmware
- run bind wizard
  - baud: 64k
  - RF power: 100
  - mode: telemetry only
- use OPLM as coordinator
- disconnect OPLM -> connect sparky2 to USB (already had latest firmware so no need to flash)
- use sparky2
- finish -> reconnect OPLM -> disconnect sparky2 USB

pug398, 2015-12-29:
I am using Sparky2+OPLM for telemetry only on TPR2 with no issues

Ga1j1n, 2015-12-30:
You are right, it seems to function on a different machine, must my be configuration

However I'm now having trouble with a OPLM Tx module I have built with a bluetooth bridge, at usable baudrates to support telemetry and PPM there are no compatible bluetooth baudrates.

Could we use more standard rates e.g 57600 or 115200 rather than 64000 or 100000

mlyle, 2015-12-30:
@ga1j1n the radio air rate (64000, 100000) can be completely distinct from the baud rate you use to talk to it.   e.g. 115200 + a radio air rate of 100000 is OK.  (You also won't get the full air-rate because of channel turnaround-time, etc)

mlyle, 2015-12-31:
@ga1j1n -- may we close this issue?

Ga1j1n, 2015-12-31:
Yes, I think so, but if I may, how do I configure the 2nd port on the OPlink to the correct baudrate for the bluetooth

mlyle, 2015-12-31:
I think it's under the module settings pane-- under misc-- you can specify the telemetry serial baud rate.

closed 2015-12-31 by mlyle

pug398, 2015-12-31:
The main port is not set for Telemetry as default in pipxtreme. You must change it to Telemetry and then save the setting. Since the uavo is not transferred to the GCS it will always be grayed out. The default pipx comspeed is set for 57k IIRC.

Ga1j1n, 2016-01-01:
Okay, when the OPLM is connected despite data being transferred the hardware tab lists no recognised board so I cannot set the main / flexi port configuration, equally modules are not available to allow me to turn on telemetry.

Do we have a different bug here?

mlyle, 2016-01-01:
@Ga1j1n It's just not very user friendly.  You need to go to the uav object browser (on the right side of configuration in most views) and look at hwtaulink --- in there you can adjust the settings and then hit the disk to store to the board.

Ga1j1n, 2016-01-01:
Thanks for the reply but I don't think the GCS is identifying the OPLM correctly, the HWTauLink settings do not match those I set in the RFMB Wizard at all and are greyed out.

The link is definitely working correctly as I can get telemetry and PPM but I don't think the settings are being correctly exposed, the available settings are more in line with a flight controller than a modem

Perhaps the UAV Settings needs an HW PipXtreme or HWOPLM setting to differentiate it or the code needs a little looking at?

mlyle, 2016-01-01:
@Ga1j1n they will be all greyed out.

Basically, the GCS greys everything out unless it has done a connection handshake.  (We should special-case this object).  Only flight boards do that connection handshake.  It still works if greyed.

The taulink/oplm/TLM/original pipxtreme hardware are all the same, supported by the same firmware, and use this same object.   You just need to plug in the OPLM, set the 'MainPort' to 'Telemetry', and it will be sending telemetry out the serial port at 57600 (when USB is not connected).  I (& @pug398 & others) have done this before.

pug398, 2016-01-02:
This has always been the case but I can see you wanting some confirmation of what is going on. Connect fc, start the gcs and bring up system>settings>HwTauLink. It will be grayed out defaults. Run the RFM bind wizard and set the fc as coordinator. Click next and connect OPLM. Upon detection of the OPLM the uavo will be transferred and you can see HwTauLink current settings in the background. Bind to coordinator and finish. Before disconnecting the OPLM you can change and save any of the settings. Do it all over again if you want to see if indeed it did take. This is the only way you will see current OPLM settings but even then they will be grayed out and not highlighted.

Ga1j1n, 2016-01-02:
Again, Thanks guys, I've followed your instructions but I still get no telemetry via bluetooth, I've followed the instructions to the letter but nothing, the link is made, PPM works, Telemetry works via USB, bluetooth connects but still no data, I'm not completely confident the Mainport telemetry setting is being correctly set but hardware wise it's perfect.

Can I suggest a change to the GCS to allow checking and changing of the Taulink settings more than once, in fact even a drop down menu to select the Mainport output during the bind wizard would do

mlyle, 2016-01-02:
Please confirm you are not powering the oplink by USB.  Serial telemetry does not work when USB is connected.

Ga1j1n, 2016-01-02:
I wish it were that simple, the OPLM is being powered by the transmitter via a 5v BEC.
I've also re-checked that Rx to Tx connections are observed and tested both ways just to be sure, I'm stumped.

pug398, 2016-01-02:
Just to double check start gcs, bring up system>settings>hwtaulink. Plug in oplm and wait for pipx to be detected. Click on rfmbind wizard and in background hwtaulink settings should populat ewith current values. Cancel wizard. (may want to set ppmport to ppm only)