d-ronin/dRonin #369: CC3D memory alarm when autotune is enabled

muchzill4, 2016-01-01:
Enabling autotune module seems to consume significant amount of memory on cc3d - the memory alert is red.

I've managed to work around that problem by disabling flexi port telemetry, which is enabled by default.

mlyle, 2016-01-01:
Yup, things are really tight on F1.  We need to make this better-- both documentation and free up a few more bytes.

gregorydulin, 2016-01-07:
I just did AutoTune on a CC3D Atom using the current preview build.  With the serial ports off and no other modules loaded, memory is green.  AutoTune gave me reasonable PID settings, and with Damping up and Noise Sensitivity down I get silky smooth flying with just a bit of jitter here and there.  Personally I blame it on my insistence on buying the cheapest hardware and not using a dedicated BEC.

Thanks!

closed 2016-01-21 by tracernz

nightflasher, 2017-07-02:
a few weeks ago i was looking for some new FC to replace all of my CC3D and found dRonin and loved it, but now i'm getting "low memory" error but haven't let autotune do it's job done. i've recently seen bad behavoirs with my attached DSMX satellite receiver which could go beside the memory error: the receiver stops working, CC3D goes into failsafe-mode (not configured => motor stop!) and the receiver is no longer bound.

is there anything i can do to free up some memory? i have no other module aktivated than the DSM on flexi port

PR #445 (2016-01-21): flight: linker script changes: sort BSS by alignment …
Saves 72 bytes of memory on Naze32, 152 bytes of memory on sparky2.

Also mark BSS noload; has no difference now (because all constituent BSS
sections are already marked noload) but a good practice for the future.
Most section scripts had this already, but chibios ones were missing it.

Fixes #14

Fixes #369

[[image]](https://reviewable.io/reviews/d-ronin/dronin/445)