d-ronin/dRonin #373: GCS error message on every startup

droidicus, 2016-01-02:
User reported that he has an error message show up every time he boots the GCS, but can continue beyond that, and the GCS still works:
> Does it matter that I get an error everytime I launch Dronin or Tau that says: "The program can't start because MSVCR120.dll is missing if I hit "OK" it launches and runs fine

[image]

Same problem exists on Tau as well.

tracernz, 2016-01-02:
MSVCR(untime)120.dll
Needs the runtime libs, e.g. https://www.microsoft.com/en-us/download/details.aspx?id=40784

mlyle, 2016-01-02:
Yup.  Which we should probably install.

droidicus, 2016-01-02:
The thing that is confusing to me is that it still runs even though the error is shown.

tracernz, 2016-01-02:
Probably one of the plugins will be broken.

droidicus, 2016-01-14:
BTW, I was able to test this on my friend's computer this weekend, and installing the MSVC runtime fixed the issue.

mlyle, 2016-01-14:
This one probably will end up as errata this time around.. almost everyone should have the runtime already

mlyle, 2016-04-01:
Errata in relnotes for Tanto.

PR #1145 (2016-07-26): Build: Windows build improvements, VS2015, Pick up QT 5.6.1
Fixes #295 
Fixes #373 
Connects to #290
Supersedes #1142
closed 2016-07-26 by tracernz