d-ronin/dRonin #390: Naze32 r.5 should support non/de/un-inverted s.bus configurations on Receiver Port Serial

fujin, 2016-01-05:
It's fairly common the miniquad world to bypass the inverter on the x4r-sb radio via the a-holes mod.

This makes the radio send "normal" s.bus, i.e., un-inverted.

It'd be nice to support this.

tracernz, 2016-01-05:
We should be careful not to call it "normal" S.BUS as normal/standard S.BUS is inverted. I think "modified S.BUS" would be better.

jihlein, 2016-01-05:
If some can test it, I'll code it.

Not sure if we have to differentiate the name just because there is no onboard inverter for the SBus signal.  It's pretty well known the Naze32 boards (at least prior to rev6 ?) need an external inverter, unless the receiver is modified.  

Longer term, I think in the wiki, each target should have "chapter" on setting it up, and that is where we could go into the details of "Naze32 requires an external Sbus inverter unless...", "Sparky 1 does not require an external Sbus inverter", etc.  

Same arguments apply to FrSky Hub telemetry wrt signal inversion.

There are so many permutations on board peripherals that its going to be hard to keep the setup nomenclature 100% clean.

Thoughts?

henn1001, 2016-01-05:
Openlrs receivers can output un-inverted sbus as well.

dustin, 2016-01-05:
@henn1001 right, and I have some and would like to use them -- even where there's hardware inverters.  That's where it gets a bit tricky.  Ideally, the port configuration bit can be explicitly optional.  naze could support ni-s.bus, and sparky could support both s.bus and ni-s.bus

mlyle, 2016-01-05:
@dustin -- that sounds ideal.  We should add a new non-inverted signalling type that doesn't do the port configuration bit.  And then add *only* that port type to Naze.  @jihlein --- sound good to you?

(I think this is also work on the way to get us factored more cleanly to not need to have quite as many args passed down to pios_hal.)

DTFUHF, 2016-01-05:
this will work great, and ni-sbus is also a great name for it, unless
something more specific is suggested.

dustin, 2016-01-05:
@mlyle I would use it on other targets since I have great hardware right now.  But it could be confusing for things like vehicle setup wizard.

tracernz, 2016-01-05:
@dustin We could add a small note to the Futaba S.BUS part of the wiring diagram SVG on targets where an external inverter or mod is required.

tracernz, 2016-01-05:
@jihlein, the point was more that if we do differentiate we should make an effort to avoid creating additional confusion by calling non-inverted S.BUS "normal". This is applicable to UI, code, and documentation.

jihlein, 2016-01-06:
I've got this coded up, it has S.Bus and S.Bus Non Inverted selections.  I converted Naze32, Sparky, and AQ32 to illustrate the setup on each processor type.  Need to do some basic testing tonight, then I'll submit a PR.  Will need some testers who have S Bus receivers to look at this further.

Nub, 2016-01-06:
@jihlein I have a quad here ready to go if you need testing, seeing I sorta spawned this ticket the other night :P

dustin, 2016-01-06:
I have a sparky and a non-inverted s.bus rx.  If we want to support this configuration on other boards, that's probably a good idea as well.  I've got Brain, quanton, cc3d, naze, and sparky for desktop testing.

mlyle, 2016-01-06:
Marking this Renatus, but it only gets in if everything goes perfectly-- nice clean minimal diff, prompt testing, etc.

pug398, 2016-01-06:
Will Brain do ni-sbus on the ppm pin? Would be nice to free up all three serial ports.

fujin, 2016-01-06:
@pug398 now that is a cool idea. I'd gladly use the a-holes mod for my brains to free up mainport for activities.

fujin, 2016-01-06:
`ni-sbus` gets a :+1: from me too.

jihlein, 2016-01-06:
@mlyle, I'll run some quick tests after work, and then create the PR.  It was a little more involved than I first thought, and a little more invasive as far the number of files touched.  From this initial PR, the boards that will support S.Bus Non Inverted operation are Naze32, AQ32, and Sparky.  Other boards will follow in a separate PR.  @dustin, @Nub, look for the PR later this evening.

As far as the Brain  being ale to do this on the PPM pin, looks like the PPM pin is PB14.  PB14 does not support uart operation, so no, the PPM pin can not support Sbus Non Inverted operation.

pug398, 2016-01-07:
Oh well it was worth asking. I was just hoping to get the main port back. Have sbus rx on 5 different fc already and can use ppm on the brain if needed :)

pug398, 2016-01-07:
@jihlein , thank you for checking on it

jihlein, 2016-01-07:
Pull Request #403 created.

PR #403 (2016-02-21): Add Sbus Non Inverted Mode
Fixes #390.

This PR implements a non-inverted Sbus interface on the following targets: AQ32, Naze32, and Sparky.  The mode is called Sbus Non Inverted on the GCS setup screens.  The 3 targets implemented here give an F1, F3, and F4 example.

I've tested the capability to selected both Sbus and Sbus Non Inverted on all applicable ports on the above 3 targets.  Each target was able to boot and execute as expected.  As I don't have any Sbus receivers, I can not test further.  Hopefully @dustin and @Nub can test with SBus receivers.  Testing should include, at a minimum [... cut]
closed 2016-02-21 by mlyle