d-ronin/dRonin #401: GCS: Vehicle Setup: Helicopter: Default curve2 should be set for CP helicopters

ekaszubski, 2016-01-06:
- Default curve2 (collective pitch) for helicopters only allows for positive collective pitch whereas it should be set to allow for the full range of collective pitch
- Specifically, the points should be [-1,-1][-0.5,-0.5][0,0][0.5,0.5][1,1] instead of [-1,0][-0.5,0.25][0,0.5][0.5.0.75][1,1]

mlyle, 2016-01-06:
Are you sure?  We should default for 3D operation?

Not sure these settings are sane with current implementations of advanced flight modes.

ekaszubski, 2016-01-06:
So having full collective range doesn't necessarily imply a 3D setup. It's often desirable to have at least some negative collective to descend quickly (going to neutral can have surprisingly little effect), perform autorotations, stick the vehicle to the ground during rotor spin-up, and increase drag on the rotor after landing to slow it down faster. 

I see three scenarios for helis:
 1. Static pitch variable throttle setup in which there is no collective channel (and thus having full collective is irrelevant, but having curve2 default to full collective will have no negative consequences)
 2. Dynamic pitch sport flying in which some negative collective is desired (throttle is either a step function for gov mode or slightly sloped as a function of collective)
 3. Dynamic pitch 3D flying in which full negative and positive collective is desired (throttle same as 2)

In the long term, I think we should add stabilization category for thrust/collective and allow users to set it to manual 2D/manual 3D operation, thus allowing for flight mode-based interpretations of desired thrust while still enabling individual stabilization modes to access the full collective range if desired.

ekaszubski, 2016-01-23:
@mlyle Any additional thoughts on this given recent discussions?

PR #522 (2016-01-26): gcs: Fix default collective curve being ignored on initial setup
When changing vehicle type to helicopter, the collective curve has its range on [0,1] even though the code clearly tries to have it default to [-1,1]. It seems as though refreshAirframeWidgetsValues() triggers some complex behavior that resets the the collective curve range to [0,1] regardless of the desired value. 

Thus, the simple fix presented here configures the collective curve range after the call to refreshAirframeWidgetsValues().

Resolves #401.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/522)
closed 2016-01-26 by mlyle