d-ronin/dRonin #406: Multirotor motors should spin when armed by default

dustin, 2016-01-08:
We suggest people do this.  We should make not doing so a conscious decision.

mlyle, 2016-01-08:
Maybe we should ask about it in the multirotor wizard.

jhitesma, 2016-01-08:
I constantly forget to set this and then wonder why mine won't arm....and am occasionally unpleasently surprised when I then bump the throttle and realize they ARE armed.  

So I'm ALL for this.  Or at least adding it to the wizard.

DTFUHF, 2016-01-08:
I'm all for this change. While messing around in this area it would be nice
to make the armed output value user-settable too. So if I want it at 1050
or 1100 it's up to me - or, a custom setting for brushed motors.

tracernz, 2016-01-08:
It is, it's the neutral value. You want to set it independently to neutral value (which should be the minimum value that causes the motors to spin)?

DTFUHF, 2016-01-08:
Had no idea it was the neutral value. I thought that was for 3D mode and or
reversible escs like a car.

So this change is about taking the checkbox on that output page and making
it enabled by default, nothing else?

mlyle, 2016-01-08:
I think it should be disabled by default, and the multirotor wizard should prompt the user (with it defaulting to true in there).

pug398, 2016-01-08:
I would prefer that myself since I never set them to spin when armed and never use the setup wizard :)

tracernz, 2016-01-10:
This will likely be bad in always armed configurations. We need to be careful with what happens if different combinations of UAVOs are changed between releases.

dustin, 2016-01-10:
Do people do always armed multirotors?

tracernz, 2016-01-10:
> Do people do always armed multirotors?

Unlikely, but the UAVO default applies to all vehicle types.

PR #465 (2016-01-21): gcs: setupwizard: enable spin while armed for multis
Fixes #406

[[image]](https://reviewable.io/reviews/d-ronin/dronin/465)
closed 2016-01-21 by tracernz