d-ronin/dRonin #408: Sparky Settings not shown in GCS

Ga1j1n, 2016-01-08:
The Sparky target flashed with preview 2 shows telemetry and movement in the flight data tab but no configuration is available in the configuration tab, it it briefly flashes up on plugin / removal but other nothing.

Tested on 2 separate windows 10 machines with the same results, works A-okay with Taulabs FW and GCS

pug398, 2016-01-08:
Did you erase settings and reboot the board?

Ga1j1n, 2016-01-08:
Sure Did, No joy

pug398, 2016-01-08:
There is an open pull request #403 that has a current test build in it. It has its own issues but you may be able to see if your particular problem is still present.

Ga1j1n, 2016-01-08:
Yep, that fixed it, glad the bug is squashed, thanks

mlyle, 2016-01-08:
That's weird-- I used sparky with TPR2.  But glad you don't see it on a current build.

closed 2016-01-08 by mlyle