d-ronin/dRonin #424: Consider seperate adjustable failsafe-timer.

henn1001, 2016-01-10:
To continue the discussion that i accidentally started in #357, it should be considered to implement a separate failsafe timer which disarms the craft after a predefined time after connection loss.

The current implementation is fine for a setup in which idle throttle means = motors off. In case of a failsafe event the motors will cut the power and after 30 seconds (default) the craft will disarm.

But i think the new conscious in the scene is to let the motors always spin when armed. See also -> #406 
With this change though, the parameter ArmingTimeout seems to be not sufficient enough to bring the craft, in case of a failsafe situation, into a safe state. The motors will just keep spinning at neutral speed until the fc disarms after 30 seconds by default (which is extremely long).

To quote the old OP wiki: 
"By consensus it is advised that for multi rotors, the safest condition is one where the throttle is cut to zero.  While the results may seem catastrophic for the vehicle (and often are), it is agreed by most that this is the safest state that can be applied to the widest set of operational circumstances.  It is better to let the copter fall to the ground, than to have it drift at hover throttle into a crowd of unsuspecting observers who generally don't realise just how dangerous a full set of spinning propellers can be.  Spinning props under a non zero throttle condition which come in to contact with grass, branches, or other objects can also cause ESCs to burn up, short out, and conceivably cause the LiPo battery to catch fire in an unknown location."

Also if something like this #228 is considered, then there will be a dilemma. You should to cut the motors immediately in case of a failsafe, but you also don´t want to accidentally disarm your craft in midair just because your throttle was too low for x seconds. This contradicts the current implementation with the ArmingTimeout parameter.

henn1001, 2016-01-10:
@DTFUHF 
I´m going to answer you here. 
Again, this only works if you use motor stop/no motors spinning at 0 throttle. If people use "always spinning" there is no option left besides setting the receiver failsafe to immediate disarm. Yeah you obviously cant recover with that. If the failsafe is handled properly in the FC, you can at least set a timer to something like 2 seconds. If the link hasn´t restored after the timer ends -> motor full stop.

mlyle, 2016-01-10:
> While the results may seem catastrophic for the vehicle (and often are), it is agreed by most that this is the safest state that can be applied to the widest set of operational circumstances. It is [...]

Yes, but neutral values are not anywhere near hover settings and are much less dangerous.  That says spinning props under a "non-zero throttle condition", but this is a zero-throttle condition.

I am really hesitant to complicate this further.  Most of our safety problems have been around misunderstanding these settings.  Just like there has been past concern with default-servo-rates and what happens to planes when actuatorsettings are reset, this is a gotcha for non-multirotor-- you don't want even fairly prolonged failsafes to disarm fixed wing.

It's likely any future implementation of air-mode will have a state machine to detect flying / recent-flying state, so I don't think we should complicate the discussion with hypotheticals about how future enhancements might work.

PR #691 (2016-03-01): flight/manualcontrol: separate failsafe-disarm timer
Defaults to off / just using the low throttle mechanism

No UI yet.  Fixes #424, but a new post-tanto issue should be opened for changing the
default and creating UI.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/691)
commit 92fd429f (2016-03-01): flight/manualcontrol: separate failsafe-disarm timer

Defaults to off / just using the low throttle mechanism

No UI yet.  Fixes #424, but a new issue should be opened for changing the
default and creating UI.
closed 2016-03-01 by tracernz