d-ronin/dRonin #425: S.Bus control safety anomaly: throttle up, unresponsive to disarm switch

fujin, 2016-01-10:
I experienced a very concerning safety event yesterday at the field where I had my goggles on, armed, applied some throttle and immediately lost control of the RC, though the OSD did not indicate failsafe (by moving into ST1).

I heard throttle starting to increase, flipped my disarm switch, the machine did not respond, throttle continued to increase. I kicked the quad off the ammo can I was launching from, held it upside down with my shoe and ripped the xt60 plug out.

I'm pulling the DVR footage. We could potentially audit s.bus failsafe path / s.bus frame receipt code paths as well.

fujin, 2016-01-11:
not sure where to put this.. this was the 'flight stats' anomaly/disarm in-flight today on shrike. 

https://www.youtube.com/watch?v=IKdDf3exeeE

pug398, 2016-01-11:
I have had zero sbus issues on 5 fc but I do not use an arming switch. I think the switch code needs serious auditing. One may crash next flight but no control problems since sbus was adopted for me.

mluessi, 2016-01-11:
It seems like the s.bus driver sets the channels to defaults if no valid s.bus frame is received for 100ms. When switch arming is used, this can mean it acts like the switch has been switched to disarm (if the channel defaults correspond to switch in disarm state). This seems to have happened in the video by @fujin, i.e., it disarms (stats are shown) and then immediately arms again when the s.bus data is valid again.

pug398, 2016-01-11:
Considering my mostly sedate flying a 0.1 second glitch every now and then could go undetected I suppose.

fujin, 2016-01-11:
https://youtu.be/hzaeqhbNsFo

mlyle, 2016-02-03:
Closing; we'll keep a record of this but there's nothing actionable here and we've made substantial sbus improvements.

closed 2016-02-03 by mlyle