d-ronin/dRonin #436: calibrate escs with test motor outputs enabled

5PAZT1C, 2016-01-14:
If you click test outputs, you can still calibrate escs with the motors spinning. Note: the motors immediatly stop spinning as soon as you click calibrate. After calibration, "test outputs" no longer functions and the output page is greyed out. 

Tested with CC3D and Brain. Neither board freezes after doing this, it's purely a GCS issue(as far as I can tell

steps to reproduce: 
check "test outputs"
[motors stop spinning]
click "calibrate escs" (it doesnt matter how many you select as long as at least one is selected)
perform esc calibration (doesnt matter whether battery is connected or not at this point)
[motors still not spinning, even with power cycle at esc]
uncheck "test outputs"
outputs page is greyed out, even if board is disconnected and reconnected; exiting GCS and re-opening it resolves the issue.

5PAZT1C, 2016-01-14:
in the same vein, if you select test outputs, then move one of the sliders, you can select another tab and you will not get a popup message asking to save your changes/abandon changes. The motor will keep spinning. 

You can also test outputs, and close the GCS and motors will continue to spin until power is removed from the esc. I will pkay around with it more once I have another battery charged. 

This has only been tested with my CC3D board (littlebee 20 escs)

mlyle, 2016-02-23:
Largely fixed by #452

closed 2016-02-23 by mlyle