d-ronin/dRonin #437: Lux: udev rule on linux missing.

mlyle, 2016-01-14:
Also ideally udev rules would go in the linux package.

closed 2016-01-14 by tracernz

reubenfirmin, 2016-02-09:
Is Renatus in the ubuntu repos yet? The Jan 26th build of Dronin still does not install udev rules.

mlyle, 2016-02-09:
@reubenfirmin No, we're not in the ubuntu repos and haven't sought inclusion-- for a couple of reasons.  First, we don't exactly conform to the recommended filesystem layout in our .deb, and second we're likely to release too often for the upstreams to keep up in a useful way (e.g. Debian/Ubuntu doesn't really like picking up feature changes after releasing, instead trying to keep the same major version and only pick up bugfixes... but we are going to release more often than Ubuntu).

How did you install?  Did you use the .tar.xz or the .deb?  The .deb is supposed to automatically install udev rules-- if it doesn't that would be nice to know.  Otherwise, from the release notes:

```
For Linux you will have to install .udev rules for the device. The .deb package attempts to do this automatically. To do this manually, download the udev permissions file and then run sudo cp _package.udev /etc/udev/rules.d/45-dronin-permissions.rules && sudo udevadm control --reload-rules
```

reubenfirmin, 2016-02-10:
@mlyle Right, I installed from the PPA. Understood about not wanting to be in the official repos. But, correct, the PPA installed deb _did not_ add udev rules.

Thanks

mlyle, 2016-02-10:
@reubenfirmin -- we don't have a PPA as far as I'm aware, just the bare deb-- did you use our bare .deb or did someone put us in their PPA somewhere?

reubenfirmin, 2016-02-10:
@mlyle Hm, my bad. But yeah, I must have used your bare .deb in that case (the 64 bit linux one.)

PR #439 (2016-01-14): Linux: improvements to udev
* Include LUX
* Include the udev rules in the compressed tar.xz.

Fixes #437

[[image]](https://reviewable.io/reviews/d-ronin/dronin/439)