d-ronin/dRonin #438: Linux deb artifact malformed

mlyle, 2016-01-14:
```
00:37 < tracernz> and in details: "Lintian check results for 
                  /home/mike/Downloads/dronin_1.d-preview-20160114-1_amd64.deb:
00:37 < tracernz> E: dronin: maintainer-address-malformed <>
00:37 < tracernz> E: dronin: maintainer-name-missing <>
00:37 < tracernz> E: dronin: maintainer-address-missing <>"
```

tracernz, 2016-01-14:
Looks like we're missing `AUTHOR` from https://github.com/d-ronin/dRonin/blob/preview-20160114/branding/linux_packaging.dict. That ends up going into https://github.com/d-ronin/dRonin/blob/preview-20160114/package/linux/deb/control#L4

tracernz, 2016-01-14:
Oops, that's wrong, we do replace the AUTHOR with this https://github.com/d-ronin/dRonin/blob/28ae7b9a3df0ebfc1106408340576ae628d48b15/package/Makefile.linux#L19

tracernz, 2016-01-14:
I guess Jenkins doesn't have a name and email set in it's git config.

tracernz, 2016-01-14:
```
mike@mike-desktop:~$ dpkg -p dronin
Package: dronin
Priority: extra
Section: unknown
Installed-Size: 152225
Maintainer: <>
Architecture: amd64
Version: 1.d-preview-20160114-1
Depends: libc6 (>= 2.14), libfontconfig1 (>= 2.9.0), libfreetype6 (>= 2.2.1), libgcc1 (>= 1:4.1.1), libgl1-mesa-glx | libgl1, libglib2.0-0 (>= 2.22.0), libice6 (>= 1:1.0.0), libpulse0 (>= 1:0.99.1), libsm6, libstdc++6 (>= 4.6), libudev1 (>= 183), libusb-1.0-0 (>= 2:1.0.9), libx11-6, libx11-xcb1, libxcb1, libxext6, libxi6 (>= 2:1.2.99.4), libxrender1, zlib1g (>= 1:1.1.4)
Size: 51546714
Description: dRonin GCS
 The dRonin project ground control station.
 A program that allows you to change settings on compatible flight controllers, create flightpaths, view controller data, create and upload picoC scripts, replay logged data, etc...
Homepage: http://dronin.org
```

mlyle, 2016-01-14:
Updated Jenkins email.  Need to see if the build in #439 fixes this.

tracernz, 2016-01-14:
Fixed.

closed 2016-01-14 by tracernz