d-ronin/dRonin #443: Nav: Home pos / alt should be set at arming, not powerup

mlyle, 2016-01-14:


pug398, 2016-01-14:
This would definitely be a good thing.

dustin, 2016-01-15:
Are these one thing?

mlyle, 2016-01-15:
Basically they should be, yes.  Home pos is slightly harder than altitude but IMO we should solve both at once.

#65 is also closely related

AlessioMorale, 2016-01-18:
In LP we used a different set of coordinates, called TakeOffLocation. The rationale being that homelocation is used as origin point the local coordinate system used by wp etc, so it is advisable to not change it on the fly.
TakeoffLocation is handled here: https://github.com/librepilot/LibrePilot/blob/next/flight/modules/ManualControl/takeofflocationhandler.c
It supports preset (saved) takeofflocation, or acquired during first or each arming.

mlyle, 2016-01-18:
This is good feedback.  Probably it should be configurable, because sometimes what's desired is nav relative to the takeoff position, and sometimes not.

cpusmith, 2016-03-17:
I have the same issue as this and the same as "Baro initialization doesn't seem correct #65". Where it's most bothersome is in the OSD when you have to keep making sure that you observe the negative altitude number, which can range from -18 meters to -65 meters, and continually minus it in your head as you're flying and watching your altitude. Not that I don't mind the metal exercise but it would be better to pay attention to other readings. (-:

If someone wants to work on this I can test on a sparky1 and Sparky2.

LitterBugs, 2018-06-07:
Been using GPS modes on my A450 with a Seppuku recently.  I agree that Home position should be set to inital arm location rather than where 3D lock is first achieved.  Maybe make a UAVO that allows  to choose how Home is set.

tracernz, 2018-06-07:
We sort of have that, but just with only two options, either preset position, or GPS lock position.
```xml
<field defaultvalue="FALSE" elements="1" name="Set" type="enum" units="">
      <description>Use preset home location when enabled, otherwise use location where GPS lock occurs. BE CAREFUL with this.</description>
      <options>
        <option>FALSE</option>
        <option>TRUE</option>
      </options>
</field>
```