d-ronin/dRonin #450: Logging: Add logging to OpenLog for other targets

henn1001, 2016-01-15:
I think @jihlein already added that option to the AQ32 and Quanton target in #16 
Would be cool if this is added for other targets as well :)

I will have a look at this later.

jihlein, 2016-01-15:
If the targets are on PIOS_HAL, should just be a matter of adding OpenLog to HwXXXX.xml.  PRs are submitted to upgrade the remaining non PIOS_HAL targets FWIW.

henn1001, 2016-01-27:
@jihlein i´ve got it working on sparky 2 but i have problems parsing the file
https://www.dropbox.com/s/ylvnz7pzrrxmkny/LOG00042.TXT?dl=0
the python script is throwing me different errors - have you had any luck with this?
Any tips?

jihlein, 2016-01-28:
@henn1001, honestly I've never had much luck with the python scripts.  What I did was download the Octave program (a freeware version of Matlab) and use the LogConvert.m file in the build/matlab folder.  It's worked fine for me.  

Find Octave at: https://www.gnu.org/software/octave/download.html

Let me know if you have any questions and/or how it works for you.

henn1001, 2016-01-28:
uhh i have matlab - i read something about that but i could not find the .m file in the source folder. I did not know i have to build them explicitly.

I´ll look into that, thanks!

henn1001, 2016-01-28:
It´s getting parced with no errors - unfortunately every struct is empty :(

jihlein, 2016-01-28:
Can you post your code changes?

henn1001, 2016-01-28:
https://github.com/henn1001/dRonin/commits/sparky2OpenLog

jihlein, 2016-01-28:
Cool, I'll take a look.

jihlein, 2016-01-28:
On a quick once over it looks correct.  

The M file can parse both GCS logs and Overo logs.  Overo logs are generated by the flight controller logging, either the spi flash or open log.  Make sure the proper argument flags are set:

% To process Overo files call with
% LogConvert(path_to_file, true)

I don't recall if not setting that argument to true will result in empty structures.

I'll dig into it more later.  One question - are you building the flight firmware and Matlab M from the same source?

henn1001, 2016-01-28:
Build from the same source/fork - yeah.

`LogConvert(path_to_file, true)`

Looks like this did the trick though  :)

`***dRonin log parser***

wrongSyncByte instances:             0
wrongMessageByte instances:          0

Completed bytes:        36 of     24938
Est. time remaining, 00h:00m:00s 
24938 records in 1.49 seconds.
>> load('LOG00042.mat')`

[image]

henn1001, 2016-01-28:
^
This really needs better usability. Something similar to the Blackboxviewer chromeapp would be awesome.

henn1001, 2016-02-08:
Ok i think this does it mostly for all the f3 and f4 targets :)
Discoveryf4 did not seem that polished/flyable so i left it out.

closed 2016-02-08 by henn1001