d-ronin/dRonin #454: GCS: failed to launch on OSX?

mlyle, 2016-01-15:
@fujin hit this in the field.  Can't be repro'd by @tgreer or @mlyle or @dustin .

Running el capitan, he gets a message like:

[image]

```
This application failed to start because it could not find or load the Qt platform plugin "cocoa".

Reinstalling the application may fix this problem.
```

dustin, 2016-01-15:
I saw he was on 10.11.1.  I'm on 10.11.2.  I did xattr -c mine before I
ran, but I don't think @tgreer did.

tgreer, 2016-01-15:
10.11.2 here. Didn't do xattr.

tracernz, 2016-01-15:
https://forums.plex.tv/discussion/186980/plex-media-server-will-damage-your-computer-you-should-move-it-to-the-trash

mlyle, 2016-01-15:
10.10.5 here, works fine.

tracernz, 2016-01-15:
10.11.2 fine. I think most likely scenario is that the DMG got corrupted.

dustin, 2016-01-15:
We verified the dmg sha1 and sha256

mlyle, 2016-01-16:
I wonder if there's a corrupt install of QT on the machine or something.

It would be helpful to get the results of running it from the command line with `export DYLD_PRINT_LIBRARIES=1`

mlyle, 2016-01-18:
@fujin a run with the above would be really appreciated as this is stop-ship if not unique to your system..

fujin, 2016-01-19:
```
aj@AJs-MacBook-Pro  ~   export DYLD_PRINT_LIBRARIES=1                                                                  14:35:12  2016-01-19
 aj@AJs-MacBook-Pro  ~   /Applications/dRonin\ GCS.app/Contents/MacOS/dRonin-GCS -r                                     14:35:23  2016-01-19
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/dRonin-GCS
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Plugins/libUtils.1.dylib
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Plugins/libExtensionSystem.1.dylib
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Plugins/libAggregation.1.dylib
dyld: loaded: /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtWidgets.framework/Versions/5/QtWidgets
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtGui.framework/Versions/5/QtGui
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtCore.framework/Versions/5/QtCore
dyld: loaded: /System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration
dyld: loaded: /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtXml.framework/Versions/5/QtXml
dyld: loaded: /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
dyld: loaded: /System/Library/Frameworks/AGL.framework/Versions/A/AGL
dyld: loaded: /usr/lib/libc++.1.dylib
dyld: loaded: /usr/lib/libSystem.B.dylib
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtSvg.framework/Versions/5/QtSvg
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtOpenGL.framework/Versions/5/QtOpenGL
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtQuick.framework/Versions/5/QtQuick
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtQml.framework/Versions/5/QtQml
dyld: loaded: /Applications/dRonin GCS.app/Contents/MacOS/../Frameworks/QtNetwork.framework/Versions/5/QtNetwork
dyld: loaded: /usr/lib/libz.1.dylib
dyld: loaded: /System/Library/Frameworks/Carbon.framework/Versions/A/Carbon
dyld: loaded: /System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa
dyld: loaded: /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
dyld: loaded: /System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices
dyld: loaded: /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
dyld: loaded: /usr/lib/libobjc.A.dylib
dyld: loaded: /System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices
dyld: loaded: /System/Library/Frameworks/Security.framework/Versions/A/Security
dyld: loaded: /usr/lib/libauto.dylib
dyld: loaded: /usr/lib/libDiagnosticMessagesClient.dylib
dyld: loaded: /usr/lib/libicucore.A.dylib
dyld: loaded: /usr/lib/libc++abi.dylib
dyld: loaded: /usr/lib/system/libcache.dylib
dyld: loaded: /usr/lib/system/libcommonCrypto.dylib
dyld: loaded: /usr/lib/system/libcompiler_rt.dylib
dyld: loaded: /usr/lib/system/libcopyfile.dylib
dyld: loaded: /usr/lib/system/libcorecrypto.dylib
dyld: loaded: /usr/lib/system/libdispatch.dylib
dyld: loaded: /usr/lib/system/libdyld.dylib
dyld: loaded: /usr/lib/system/libkeymgr.dylib
dyld: loaded: /usr/lib/system/liblaunch.dylib
dyld: loaded: /usr/lib/system/libmacho.dylib
dyld: loaded: /usr/lib/system/libquarantine.dylib
dyld: loaded: /usr/lib/system/libremovefile.dylib
dyld: loaded: /usr/lib/system/libsystem_asl.dylib
dyld: loaded: /usr/lib/system/libsystem_blocks.dylib
dyld: loaded: /usr/lib/system/libsystem_c.dylib
dyld: loaded: /usr/lib/system/libsystem_configuration.dylib
dyld: loaded: /usr/lib/system/libsystem_coreservices.dylib
dyld: loaded: /usr/lib/system/libsystem_coretls.dylib
dyld: loaded: /usr/lib/system/libsystem_dnssd.dylib
dyld: loaded: /usr/lib/system/libsystem_info.dylib
dyld: loaded: /usr/lib/system/libsystem_kernel.dylib
dyld: loaded: /usr/lib/system/libsystem_m.dylib
dyld: loaded: /usr/lib/system/libsystem_malloc.dylib
dyld: loaded: /usr/lib/system/libsystem_network.dylib
dyld: loaded: /usr/lib/system/libsystem_networkextension.dylib
dyld: loaded: /usr/lib/system/libsystem_notify.dylib
dyld: loaded: /usr/lib/system/libsystem_platform.dylib
dyld: loaded: /usr/lib/system/libsystem_pthread.dylib
dyld: loaded: /usr/lib/system/libsystem_sandbox.dylib
dyld: loaded: /usr/lib/system/libsystem_secinit.dylib
dyld: loaded: /usr/lib/system/libsystem_trace.dylib
dyld: loaded: /usr/lib/system/libunc.dylib
dyld: loaded: /usr/lib/system/libunwind.dylib
dyld: loaded: /usr/lib/system/libxpc.dylib
[... 113 lines cut ...]
dyld: loaded: /System/Library/PrivateFrameworks/IconServices.framework/Versions/A/IconServices
dyld: loaded: /System/Library/PrivateFrameworks/Apple80211.framework/Versions/A/Apple80211
dyld: loaded: /System/Library/Frameworks/CoreWLAN.framework/Versions/A/CoreWLAN
dyld: loaded: /System/Library/Frameworks/IOBluetooth.framework/Versions/A/IOBluetooth
dyld: loaded: /System/Library/PrivateFrameworks/CoreWiFi.framework/Versions/A/CoreWiFi
dyld: loaded: /System/Library/Frameworks/CoreBluetooth.framework/Versions/A/CoreBluetooth
dyld: loaded: /System/Library/PrivateFrameworks/ChunkingLibrary.framework/Versions/A/ChunkingLibrary
dyld: loaded: /System/Library/PrivateFrameworks/DebugSymbols.framework/Versions/A/DebugSymbols
[... cut ...]
dyld: loaded: /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/Print.framework/Versions/A/Print
dyld: loaded: /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/SecurityHI.framework/Versions/A/SecurityHI
dyld: loaded: /System/Library/PrivateFrameworks/CoreServicesInternal.framework/Versions/A/CoreServicesInternal
This application failed to start because it could not find or load the Qt platform plugin "cocoa".

Reinstalling the application may fix this problem.
[1]    2862 abort      /Applications/dRonin\ GCS.app/Contents/MacOS/dRonin-GCS -r
```

@mlyle

tracernz, 2016-01-19:
Mine (on 10.11.2 as well): https://gist.github.com/tracernz/1f6a74489888210b15b9

tracernz, 2016-01-19:
Michaels-MacBook-Pro:~ mike$ otool -L /Applications/dRonin\ GCS.app/Contents/Plugins/platforms/libqcocoa.dylib > ~/Desktop/otool.txt
```
/Applications/dRonin GCS.app/Contents/Plugins/platforms/libqcocoa.dylib:
	libqcocoa.dylib (compatibility version 0.0.0, current version 0.0.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 21.0.0)
	/System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 157.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/usr/lib/libcups.2.dylib (compatibility version 2.0.0, current version 2.11.0)
	@rpath/QtGui.framework/Versions/5/QtGui (compatibility version 5.5.0, current version 5.5.0)
	@rpath/QtCore.framework/Versions/5/QtCore (compatibility version 5.5.0, current version 5.5.0)
	/System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
	@rpath/QtDBus.framework/Versions/5/QtDBus (compatibility version 5.5.0, current version 5.5.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.5)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices (compatibility version 1.0.0, current version 48.0.0)
	/System/Library/Frameworks/AGL.framework/Versions/A/AGL (compatibility version 1.0.0, current version 1.0.0)
	@rpath/QtPrintSupport.framework/Versions/5/QtPrintSupport (compatibility version 5.5.0, current version 5.5.0)
	@rpath/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.5.0, current version 5.5.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1343.14.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1151.16.0)
	/System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices (compatibility version 1.0.0, current version 62.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1151.16.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
```

fujin, 2016-01-19:
reinstalled laptop. works ok now. /shrug

closed 2016-01-19 by fujin