d-ronin/dRonin #459: X4R-SB latest OEM firmware is doing something funky with HPWM RSSI

fujin, 2016-01-17:
@tracernz @mluessi have been in discussions. I am able to reproduce on brand new x4r-sb on Shrike. (always shows zero)

mlyle, 2016-01-17:
Sticking in Renatus --- just for near term visibility

fujin, 2016-01-17:
thanks, @mlyle

pug398, 2016-01-20:
@fujin On X4R you are taking rssi off the pad right below antenna?

fujin, 2016-01-20:
@pug398 sure are

pug398, 2016-01-20:
I am getting nothing but a high level signal off that pad at the moment no pwm. I will try another X4R to see if I get same. It may be because I am also routing rssi through sbus. I wanted to put a lpf on the pwm rssi out to see if it looked suitable for ADC input.

pug398, 2016-01-22:
The older X4RSB is outputting pwm rssi off the pad with same setting so I am 50/50 whether it is a change in newer OEM firmware. Need to check if newer one has been flashed with CPPM bind option.

mlyle, 2016-09-08:
Brain has moved to using ADC for this; punting.

closed 2016-09-08 by mlyle