d-ronin/dRonin #461: GCS flash log download interrupted

droidicus, 2016-01-17:
Frequently when downloading a flash log from my Sparky2 on Windows 10, the GCS will not complete the download, and it will become stuck with a half downloaded log. The Status still says "Downloading", and not "Download Complete". Restarting the GCS and the FC will let me attempt to re-download, at which point it may or may not succeed, potentially failing at a different point than the previous time. Sometimes it takes several cycles of this to successfully download a log file, especially a large/long log file.

mlyle, 2016-02-11:
Looks like there is absolutely no attempt made to retry or anything, which isn't awesome.

henn1001, 2016-02-29:
Happens to my f3 (no v-sense pin) target every time  over v-usb.
I can download it fine over serial telemetry.

I also did a quick test on my sparky2 and can confirm that the same happens there with larger files. Seems to work better with smaller files though.

I´m on win10 as well.

mlyle, 2016-03-24:
Won't be further improving this before samsara; deferring.

mlyle, 2017-06-28:
Needs to move to use file transfer infrastructure... that said I bet this is **way** better. :P

mlyle, 2018-02-04:
Both this is used much less, and it's much more reliable.  So the incentive to move it to "good" infrastructure is lessening.  deferrin'.

closed 2018-02-04 by mlyle