d-ronin/dRonin #462: GCS no longer displaying the file dialog box

droidicus, 2016-01-17:
After using the GCS for a while (5-10 minutes usually), the GCS will no longer open a file dialog box when it is needed (log file save, UAVO import/export, firmware file selection, etc). Closing and restarting the GCS fixes the issue. On Windows 10, this happens frequently, but seemingly randomly.

mlyle, 2016-01-17:
Is this a duplicate of #236?

droidicus, 2016-01-17:
I don't believe so, I am not seeing a window in front or behind.

mlyle, 2016-01-17:
So you've moved the window around and looked for it and can't find it?

droidicus, 2016-01-18:
90% sure at least.

droidicus, 2016-01-18:
Reproduced, no window is shown above or below. None on taskbar, nor shown in an alt-tab.

tracernz, 2016-01-18:
Can't repro on Lunix. The file dialogs do start opening underneath the main window after a while though, may be related.

droidicus, 2016-01-18:
Not to muddy the waters, but another possibly related issue... Sometimes I can successfully get to the dialog box, but it will not let me open or save a file. I can click the open/save button, highlight the filename and hit enter, etc. No dice.

mlyle, 2016-06-25:
@droidicus Have you seen this at all lately?  Is it still present?  I cannot repro / am inclined to close

tracernz, 2016-06-25:
I haven't seen it recently because I added that code to auto-load the correct firmware back when we started on upgrader work. It probably still happens when saving partitions, log files etc. though.

mlyle, 2016-06-25:
@tracernz-- this isn't the pop-behind one.  This is a completely missing dialog.  do you mean completely-missing or pop-behind?

tracernz, 2016-06-25:
Oh yep, you're right, never seen this one.

closed 2016-07-22 by mlyle