d-ronin/dRonin #470: Add opt-in for usage statistic and sharing of private data

mluessi, 2016-01-18:
The GCS currently silently publishes usage statistics, including private information like the IP address and flight controller serial number. IMO, this is not acceptable and we should require users to opt-in. The easiest solution would be to add a pop-up dialog asking users for permission with check boxes defaulting to true.

dustin, 2016-01-18:
Mangling serial numbers is useful.  It's not possible to not share IP addresses, though.

mluessi, 2016-01-18:
Yeah, if you submit something to a server, you are sharing the IP anyways. The important thing is to ask user for permission.

mluessi, 2016-01-18:
I just noticed that the settings are also not saved, i.e., restarting the GCS enables sharing again if it was disabled.

PR #475 (2016-01-19): GCS Usage Stats: Notify user, save config
Tested on Linux. 

Fixes #470

[[image]](https://reviewable.io/reviews/d-ronin/dronin/475)
closed 2016-01-19 by mlyle