d-ronin/dRonin #471: Request for comment: change code pattern

mlyle, 2016-01-18:
Very frequently with UAVOs, we do something like:

StabilizationDesiredGet(&stabDesired);
stabDesired.Throttle=...
stabDesired.Whatever=...
StabilizationDesiredSet(&stabDesired);

I think if we intend to set all fields, we should default or bracket initialize things instead.  This way we're not tainted by past states, we don't contend the giant uavo lock, and we don't copy data as much.

Thoughts?

droidicus, 2016-01-19:
One issue I could see here is if new variables get added to the UAVO, then that code block would no longer be setting all fields, and some data could be lost. Could be nasty to track down too.

mlyle, 2016-01-19:
UAVOs are supposed to have one producer and one consumer at all times, and flight code isn't supposed to set settings objects in general (the big exception being the systemident object).

This would be for things like stabilizationdesired, etc.  Far better things go to defaults than something some other module that was driving the UAVO before set.

tracernz, 2016-01-19:
We could turn on GCC warnings to tell us about uninitialized fields.

mluessi, 2016-01-20:
Another option could be to have the UAVO generator create set functions that just set all fields of the UAVO struct. The functions would be inlined (so be efficient) and force the programmer to set all fields.

droidicus, 2016-01-20:
@mluessi I was thinking about something like that as well, +1 from me

mluessi, 2016-01-31:
@mlyle WDYT? Lets figure this out and then go through the code and fix the offending places.

mlyle, 2016-01-31:
Yup, I think we should do it and your suggestion is a good one-- if we put it in the header so it was inlined, the redundant setters (the things we set ourselves jsut after) would be omitted too.

mlyle, 2016-02-01:
#139 should merge; then we can factor some of the initialization impl out as a next step towards this.

PR #944 (2016-04-30): flight: avoid getting uavos unnecessarily
Largely fixes #471.

Avoids copying lots of data and removes some pressure on UAVO lock.
commit f1a2666f (2016-04-30): flight: avoid getting uavos unnecessarily

Largely fixes #471.

Avoids copying lots of data and removes some pressure on UAVO lock.
closed 2016-04-30 by tracernz