d-ronin/dRonin #472: Corrupt flash log files

droidicus, 2016-01-18:
Occasionally I will get a log file from the the flash download that is not parsable from the Matlab LogConvert.m script. The problem always occurs at the end of the file. Attached is a log that is demonstrating this issue. Attached is a zip containing a log file, and the matlab script that demonstrates this. [autotuneLogsCorrupt.zip](https://github.com/d-ronin/dRonin/files/95006/autotuneLogsCorrupt.zip)

The output generated by Matlab is as follows:
```
>> LogConvert('ShortFlyNoiseDefaults.drlog' , true)

***Tau Labs log parser***

warning: Git hashes do not match. Incorrect file importer.
warning: called from
    LogConvert at line 2399 column 2
warning: implicit conversion from numeric to char
wrongSyncByte instances:             0
wrongMessageByte instances:          0

Completed bytes:        36 of    221085
Est. time remaining, 00h:00m:07s
warning: implicit conversion from numeric to char
wrongSyncByte instances:             0
wrongMessageByte instances:          0

Completed bytes:     50055 of    221085
Est. time remaining, 00h:00m:03s
warning: implicit conversion from numeric to char
wrongSyncByte instances:             0
wrongMessageByte instances:          0

Completed bytes:    100078 of    221085
Est. time remaining, 00h:00m:03s
warning: implicit conversion from numeric to char
wrongSyncByte instances:             0
wrongMessageByte instances:          0

Completed bytes:    150079 of    221085
Est. time remaining, 00h:00m:02s
warning: implicit conversion from numeric to char
wrongSyncByte instances:             0
wrongMessageByte instances:          0

Completed bytes:    200081 of    221085
Est. time remaining, 00h:00m:01s
warning: implicit conversion from numeric to char
wrongSyncByte instances:            20
wrongMessageByte instances:          0

Completed bytes:    221088 of    221085
Est. time remaining, -1h:59m:60s
error: LogConvert: A(I): index out of bounds; value 221086 out of bound 221085
error: called from
    LogConvert at line 5219 column 20
```

If line 5219 of LogConvert.m is commented out, then the file will successfully parse.

mlyle, 2016-02-07:
Matlab should be able to deal with this-- it's just truncation (trying to read past the end / get the last record which isn't fully there.)

mlyle, 2016-02-11:
Just to note, the "fixes #..." needs to be in the top comment (edited in after is OK) of the PR for autoclose to work.

closed 2016-02-11 by mlyle

droidicus, 2016-02-11:
Ah, good to know!