d-ronin/dRonin #476: Crashreporter doesn't work on OSX currently

mlyle, 2016-01-19:
```
Dump file was written to: /var/folders/kb/g2m5j1f576l3z7lk3ypyq6hr0000gn/T/dr__HEAD_detached_at_e443eaa_20160119_e443eaa5/3AAC6AC1-435D-4C05-8AFB-4A3FBB541C69.dmp
dyld: Library not loaded: @rpath/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Applications/dRonin GCS.app/Contents/MacOS/crashreporterapp
  Reason: image not found
```

mlyle, 2016-01-19:
```
Michaels-MBP:MacOS mlyle$ otool -L crashreporterapp 
crashreporterapp:
	@rpath/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.5.0, current version 5.5.0)
	@rpath/QtGui.framework/Versions/5/QtGui (compatibility version 5.5.0, current version 5.5.0)
	@rpath/QtCore.framework/Versions/5/QtCore (compatibility version 5.5.0, current version 5.5.0)
	/System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	@rpath/QtNetwork.framework/Versions/5/QtNetwork (compatibility version 5.5.0, current version 5.5.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/AGL.framework/Versions/A/AGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
```

vs.

```
dRonin-GCS:
	@executable_path/../Plugins/libUtils.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	@executable_path/../Plugins/libExtensionSystem.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	@executable_path/../Plugins/libAggregation.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1256.14.0)
	@rpath/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.5.0, current version 5.5.0)
	@rpath/QtGui.framework/Versions/5/QtGui (compatibility version 5.5.0, current version 5.5.0)
	@rpath/QtCore.framework/Versions/5/QtCore (compatibility version 5.5.0, current version 5.5.0)
	/System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	@rpath/QtXml.framework/Versions/5/QtXml (compatibility version 5.5.0, current version 5.5.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/AGL.framework/Versions/A/AGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
```

(AKA looks sane).

jpbarraca, 2016-02-04:
I made GCS crash after the crashreporter was initialised and it was called correctly. Got the popup and and could save the report. Using githash b233e9ed

OS X El Capitan 10.11.3

mlyle, 2016-02-04:
@jpbarraca Thanks.

Unfortunately I think it works when you build it yourself, but not in packages.  I need to look into why.

jpbarraca, 2016-02-05:
Confirmed when simply compiling GCS and when using GCS from a locally built package.

mlyle, 2016-07-13:
I suspect this is the critical difference:

dronin-gcs:
```
otool -l dRonin-gcs
[...]
Load command 26
          cmd LC_RPATH
      cmdsize 48
         path @executable_path/../Frameworks (offset 12)
```
vs  crashreporter:
```
Load command 20
          cmd LC_RPATH
      cmdsize 136
         path /Users/Shared/Jenkins/Library/Developer/org.jenkins-ci.slave.jnlp/workspace/dronin/Nodes/osx/tools/Qt5.6.0/5.6/clang_64/lib (offset 12)
```

mlyle, 2016-07-13:
I suspect these lines need to be added to crashreporterapp.pro:

```
macx {
        QMAKE_LFLAGS_SONAME = -Wl,-install_name,@executable_path/../Plugins/
}   
```

mlyle, 2016-07-13:
No, that's not it/enough.

mlyle, 2016-07-14:
Here it is:

> macdeployqt supports the following options:  
> ...  
> -executable=<path>	Let the given executable also use the deployed frameworks

Should probably be added to the arguments in package/osx/libraries for crashreporter.

PR #1110 (2016-07-14): Breakpad: Use binary dist., get all symbol info on OSX
Binary dist. is hosted by me at the moment which we may like to change. That can happen after Samsara though, need to get this on Jenkins ASAP.

Now gets all the symbol info and unmangles symbols on OSX.

Need to test that everything works on Linux. Would be nice to get Windows also but I don't have a working Windows build env. ATM.

To build package this environment var. needs set: `EXPORT_SYMBOLS=YES`

I might add something to zip up the symbols package so we can collect them easily with the Jenkins artifacts thing. (-e- package_all_comp [... cut]
closed 2016-07-14 by mlyle