d-ronin/dRonin #482: Fix VirtualBar documentation

dustin, 2016-01-20:
This is a mostly-heli feature, but the docs imply it has benefits for multirotors.  We should be clear about what this is and where you should use it.

raemin, 2016-01-22:
Just updated the wiki with the appropriate caveats. Should anyone point me to the correct direction (deprecated taulabs doc, or whatever could be usefull), I'll hapilly test it and report. Maybe this flightmode would also benefit from a proper tuning pane in the GCS... This would kind of male the documentation pointless.

mlyle, 2016-04-01:
I skimmed docs and made a couple minor edits.  Looks good.  I don't love the separate PIDs but I see why someone would want different values.  Closing.

closed 2016-04-01 by mlyle