d-ronin/dRonin #487: RFC: Multiplex SRXL/UDI/XBus Mode.B RX driver

tracernz, 2016-01-21:
Does anybody/would anybody use a driver? I wrote one last night but I'm not sure if anybody would use it. It needs a little cleanup and then I can submit a PR if it's wanted.

tracernz, 2016-01-21:
Supported by JR, Jeti, Multiplex, and possibly some other radios. I have tested framerates between 5 ms and 30 ms.

tracernz, 2016-01-21:
The diff: https://github.com/d-ronin/dRonin/compare/next...tracernz:srxl?expand=1

mlyle, 2016-01-21:
How much code size does it use?

tracernz, 2016-01-21:
next:
```
/home/mike/Dev/dronin/build/fw_sparky/fw_sparky.elf  :
section                   size        addr
startup                    392   134266880
.text                   209312   134267296
_exti                       44   134476608
.initcalluavobj.init         0   134476652
.initcallmodule.init       208   134476652
.ARM.exidx                   8   134476860
.textalign                   4   134476868
.boardinfo                   0   134233984
.stacks                   2048   268435456
.ccm                        56   268437504
.fastheap                 6088   268437560
.data                     1736   536870912
.bss                      8800   536872648
.heap                    30424   536881448
.debug_info             727852           0
.debug_abbrev           139895           0
.debug_aranges           27336           0
.debug_ranges            30776           0
.debug_line             278006           0
.debug_str              228045           0
.comment                   112           0
.ARM.attributes             55           0
.debug_frame             69708           0
.debug_loc              279829           0
.debug_macro             32709           0
Total                  2073443
```
srxl:
```
/home/mike/Dev/dronin/build/fw_sparky/fw_sparky.elf  :
section                   size        addr
startup                    392   134266880
.text                   209936   134267296
_exti                       44   134477232
.initcalluavobj.init         0   134477276
.initcallmodule.init       208   134477276
.ARM.exidx                   8   134477484
.textalign                   4   134477492
.boardinfo                   0   134233984
.stacks                   2048   268435456
.ccm                        56   268437504
.fastheap                 6088   268437560
.data                     1736   536870912
.bss                      8804   536872648
.heap                    30420   536881452
.debug_info             730387           0
.debug_abbrev           140714           0
.debug_aranges           27400           0
.debug_ranges            30880           0
.debug_line             278833           0
.debug_str              228650           0
.comment                   112           0
.ARM.attributes             55           0
.debug_frame             69884           0
.debug_loc              281197           0
.debug_macro             32709           0
Total                  2080565
```

624 bytes extra in .text

mlyle, 2016-01-21:
Fine with me, let's leave it off F1 and bring it elsewhere.

m-thread, 2016-01-21:
This looks good. I already did this in OP/LP quite some time ago. Could we not try to collaborate around this instead of inventing the wheel over and over again?

tracernz, 2016-01-21:
Yes we should. In this case I wrote the driver in a couple of hours for my own benefit, I wasn't even planning to do a PR. ;-) On that note, we should discuss the Jeti driver. :-D Are you on IRC at all?

tracernz, 2016-01-21:
Unfortunately I can't figure out how to see your PRs to see what's going on at LP. I'm too used to Github being so easy. :-P

tracernz, 2016-02-15:
Turns out HoTT SUMD/H is a variant of SRXL, would be good to incorporate it into the SRXL driver to avoid code duplication.

m-thread, 2016-02-15:
Yes. I think it would be easy to have the SUMD/SUMH to be auto detected as well. No need to configure one of the two types when it can be deduced from the stream?

mlyle, 2016-06-25:
With the SRXL work so far, time to bump to Quixote

tracernz, 2016-06-25:
Remaining things to do:
* [ ] Move SUMD into pios_srxl
* [ ] Move SUMH into pios_srxl
* [ ] Cleanup setup wizard etc. as required