d-ronin/dRonin #495: Push

ekaszubski, 2016-01-22:


tgreer, 2016-01-22:
I think this is a good idea. Manual is a terrible terrible thing, unless you REALLY know what you're doing.

mlyle, 2016-01-22:
I don't think it should be removed entirely.  There's existing fixed-wing setups that this breaks.  We already prohibit it entirely on multirotor.

ekaszubski, 2016-01-22:
@mlyle I went through all the references to manual flight mode; selecting any stabilized1/2/3 flight mode and setting all the stabilization axes to "manual" should result in identical functionality as selecting manual flight mode.

mlyle, 2016-01-22:
Sure.  But it breaks existing configurations.  By the same logic we could remove most of the flight modes because you can get equivalent behavior by using the stabilization banks.  But that confuses manypeople and in any event breaks people who have existing configuration.

ekaszubski, 2016-01-22:
IMO having a redundant flight mode that can also specified per-axis via stabilization modes is more confusing (similar confusion led to a crash for me and I've been using the GCS for a while). If we keep the redundant flight modes, I think any unused stabilization banks should be hidden or grayed-out in the GCS.

As for breaking existing configurations, a minority of users will be affected. How do we decide when it's OK to break existing configurations, or are you saying we need to provide indefinite legacy support for existing configurations?

pug398, 2016-01-22:
I would ask is there a supported vehicle type that uses "manual" as a primary flight mode or does just about everyone (for that vehicle type) have to go in to stabilize and set custom modes per axis?

tracernz, 2016-01-22:
The only vehicle type that doesn't use manual is multirotor so it might be wise to keep it around.

DTFUHF, 2016-01-22:
I'd say manual mode is required for airplanes, I'd never want to fly an
airplane with an autopilot that didn't have a manual mode I could switch to.

ekaszubski, 2016-01-22:
To clarify, I'm not suggesting we remove full-manual functionality, I'm just saying we should push that functionality into stabilized1/2/3 banks (so make it a stabilization mode instead of a flight mode) so that it can be enabled on individual (or all) stabilization axes. Then you can just set stabilized3 to manual/manual/manual for roll/pitch/yaw and you get the same functionality as selecting manual flight mode.

However, @mlyle makes a good point that we should try to avoid breaking things for people unless there's some significant benefit, so I don't see any significant issue in keeping the manual flight mode around at the cost of slightly more complex code (to handle both manual flight mode and manual stabilization mode per-axis).

For completeness, these are the cases I can see for manual/mixed/stabilized modes:
- vehicles that should not use *any* manual stabilization except for bench testing:
  - multirotors
  - flybarless helicopters
- vehicles that should use *some* manual stabilization (we currently do not support these configurations):
  - some fixed-wing (any axis *may* or *may not* be stabilized via an IMU)
  - most flybar helicopters (manual roll and pitch; yaw *may* or *may not* be stabilized via gyro)
  - some ground vehicles (any axis *may* or *may not* be stabilized via a variety of methods including encoders, IMU, cameras, etc; differential-drive vehicle: rotate at 30 degrees per second; drive forward at 0.5 meters per second; not sure if we support either of these for ground anyway)
- vehicles that should use *full* manual stabilization:
  - some fixed-wing
  - few flybar helicopters (throttle+collective->yaw mixing used instead)
  - most ground vehicles

mlyle, 2016-01-22:
> so I don't see any significant issue in keeping the manual flight mode around at the cost of slightly more complex code (to handle both manual flight mode and manual stabilization mode per-axis).

IMO it should just be like the other ones-- using 'manual' pushes each of the 3 axes into manual modes.

ekaszubski, 2016-01-27:
> IMO it should just be like the other ones-- using 'manual' pushes each of the 3 axes into manual modes.

Agreed, and this is how I implemented it. Ready for review: #496

PR #496 (2016-01-31): Removed Manual flight mode; added Manual stabilization mode
Fixes #495 
- Updated sanity checks
- Pushed transmitter -> actuator logic into stabilization
- Updated/removed references to manual mode in:
  - OnScreenDisplay
  - UAVOHoTTBridge
  - UAVOLighttelemetryBridge
  - UAVOMavlinkBridge
  - Various GCS plugins

[[image]](https://reviewable.io/reviews/d-ronin/dronin/496)
closed 2016-01-31 by mlyle