d-ronin/dRonin #513: gcs: curve2 source always shows "Throttle"

ekaszubski, 2016-01-23:
Regardless of actual curve2 source, the graph widget in gcs always says that the source of curve2 is "Throttle"

tracernz, 2016-03-20:
Where are you looking?

ekaszubski, 2016-03-21:
Configuration -> Vehicle -> Mixer Settings [Helicopter] -> Curve settings -> below curve2 graph

Note that MixerSettings.Curve2Source in the property manager says "Collective" while the label on the graph says "Throttle"

[image]

tracernz, 2016-03-21:
Thanks, will take a look later.

ekaszubski, 2016-03-21:
To clarify, the GCS in the above screenshot was built from March 17 next (6bee0b131ee6870e1cfcdbda8cd7dcc9b311317a)

mlyle, 2016-03-24:
Won't be further improving this before samsara; deferring.

mlyle, 2016-09-01:
It looks like this is half implemented.  mixercurve.cpp has plumbing for 'CBCurve2Source'; configccpmwidget.cpp does `m_ccpm->PitchCurve->setMixerType(MixerCurve::MIXERCURVE_OTHER);`  (there's a flag it doesn't set to enable that widget.

But the value doesn't seem to get stored/restored from uavos, etc.

PR #2195 (2018-05-09): gcs: Fix Curve2 UI.
The curve 2 widget was never told it's actually curve 2, so it hid the dropdown and showed a label instead.

Also, the dropdown was never hooked up to the MixerSettings.Curve2Source to begin with.

Should deal with #513?

Fixes #513
closed 2018-05-09 by mlyle