d-ronin/dRonin #515: Consolidate Throttle/Collective into "Thrust" or "Power" axis for stabilization pipeline

ekaszubski, 2016-01-23:
Issues:
- If setup is cpheli, "thrust" is commanded via ManualControlCommand.Collective
- If setup is multirotor, "thrust" is commanded via ManualControlCommand.Throttle
- All stabilization methods for altitude have their climb rate set exclusively from ManualControlCommand.Throttle; thus, they will produce invalid behavior for cpheli setups since Collective is the expected control source, not Throttle
- Similarly, the "thrust" component of StabilizationDesired is derived from ManualControlCommand.Throttle, which will produce invalid behavior for cpheli
- The Acuator module prevents any autonomous control of the collective channel by checking curve2 source and mixing Collective via ManualControlCommand.Collective passed through curve2, so Collective can only be set directly as a stick value or indirectly as a function of one of the roll/pitch/yaw axes (the latter is advanced/probably bogus behavior)

Proposed solution:
- Actuator module should:
  - If multirotor/fixed-wing/etc -> mix Throttle via ActuatorDesired.Throttle passed through curve1 (current behavior)
  - If cpheli -> mix Collective via ActuatorDesired.Throttle passed through curve2, and mix Throttle via ManualControlCommand.Throttle passed through curve1
- The Throttle field should be renamed to Thrust or Power where applicable (StabilizationDesired, ActuatorDesired, LoiterCommand, etc) to better convey the value it represents
- The ManualControl module should select between ManualControlCommand.Throttle and ManualControlCommand.Collective as a source for the Thrust/Power axis depending on vehicle type

tracernz, 2016-01-25:
Sounds like a good proposal, PR welcome.

ekaszubski, 2016-01-25:
@tracernz @mlyle Preferences on new name for axis? Thrust, Power, something else? Thrust is my preference but it doesn't make a lot of sense for ground vehicles.

mlyle, 2016-01-25:
I like thrust.

ekaszubski, 2016-01-29:
@tracernz @mlyle Looks like shortest path to selecting the "Thrust" source is:
- If SystemSettings.AirframeType == HeliCP --> use collective
- else --> use throttle

However, fixed-pitch helicopters still expect to use throttle; thus it seems like we should add a new AirframeType called "HeliFP" or something similar (above check would still select correct thrust source for this situation).

tracernz, 2016-01-29:
Do we even want to support fixed pitch? Surely nobody serious enough to fit an aftermarket controller is going to fly FP.

ekaszubski, 2016-01-29:
@tracernz I mean I'm fine with not supporting FP, just trying to cover all the bases in the planning phase.

PR #569 (2016-02-09): Consolidated Throttle/Collective into Thrust
Resolves #515.

Updated field names in UAVOs:
- ActuatorDesired
- AltitudeHoldState
- LoiterCommand
- StabilizationDesired

Updated references and throttle/collective logic in various flight modules.

This PR also enables helicopters to use flight modes which stabilize altitude.

[[image]](https://reviewable.io/reviews/d-ronin/dronin/569)
closed 2016-02-09 by mlyle