d-ronin/dRonin #518: CC3D/DiscoveryF4: Compiling with ERASE_CHIP=YES broken

tracernz, 2016-01-24:
Broken in 2013 by @solidgoldbomb in https://github.com/d-ronin/dRonin/commit/ac53e9d518c94c3d785eeed8466c1cd043c291f8#diff-c64214d8dc83380dce317d7c0aa66a18.
We should remove this on the Disco since F4 has partition manager so it's unnecessary. Need to write some new code to erase all the sectors on CC3D since it doesn't have a partition manager so there's no way to erase settings that cause a reboot loop.

mlyle, 2016-01-24:
ERASE_FLASH=YES for clarity

Would be nice to remove it from all F3/F4 targets.

I notice it looks like this on F3/F4:

```
#if defined(ERASE_FLASH)
        PIOS_FLASHFS_Format(pios_uavo_settings_fs_id);
#endif
```

Would this work?

tracernz, 2016-01-24:
Yeah, that's probably the right thing. I'm guessing the code on CC3D was from before the new filesystem was implemented.

PR #528 (2016-01-28): Move discoveryf4/coptercontrol to new flash erase paradigm
Fixes #518

[[image]](https://reviewable.io/reviews/d-ronin/dronin/528)
closed 2016-01-28 by tracernz