d-ronin/dRonin #527: Sparky: Attempt to free some flash

tracernz, 2016-01-24:
So new features can be added. Best candidates:
```
Michaels-MacBook-Pro:dronin mike$ ./tools/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-nm -S --size-sort --reverse-sort --radix=dec ./build/fw_sparky/fw_sparky.elf 
134359376 00013766 T CovariancePrediction
134454192 00012288 T twiddleCoefQ15
134308400 00005308 t AutotuneTask
134335856 00003068 t uavoHoTTBridgeTask
134296432 00002796 t stabilizationTask
134283824 00002264 t updateAttitudeComplementary
134471168 00002184 t CoeffFile
134292896 00002084 T transmitter_control_update
134324368 00002076 t pathfollowerTask
134452144 00002048 T armBitRevTable
134313856 00002032 t updatePIDs
134281856 00001964 t updateAttitudeINSGPS
134339248 00001932 t uavoFrSKYSensorHubBridgeTask
134407824 00001796 t PIOS_MPU9150_Task
134446864 00001520 T __ieee754_powf
134449488 00001506 T __kernel_rem_pio2f
134328096 00001448 t uavoMavlinkBridgeTask
134278960 00001444 t SensorsTask
134344256 00001288 T PIOS_Board_Init
134329952 00001252 t uavoMSPBridgeTask
134333520 00001218 t generate_warning
134317856 00001216 t pathPlannerTask
134304320 00001164 T parse_ubx_stream
134386432 00001152 T PIOS_INTERNAL_ADC_Init
134306736 00001148 t attitudeUpdated
536871576 00001064 d impure_data
134286096 00001052 t AttitudeTask
```

mlyle, 2016-01-24:
> 134452144 00002048 T armBitRevTable

This is vibration analysis I think.

> 134335856 00003068 t uavoHoTTBridgeTask

Why is this so much bigger than all the other serial rx protocol things?

Most of the rest is fairly necessary INS / etc stuff.

tracernz, 2016-01-24:
Yeah but we might be able to squash some of it down without removing it.

PR #595 (2016-02-04): sparky: remove vibration analysis
We're just way too down to the wire on flash space on Sparky1, and this
is a very infrequently used feature that uses a ton of space.

before:

`-rw-r--r--  1 mlyle  staff  211268 Feb  2 11:21 build/fw_sparky/fw_sparky.tlfw`

after:

`-rw-r--r--  1 mlyle  staff  193004 Feb  2 11:20 build/fw_sparky/fw_sparky.tlfw`

fixes #527

[[image]](https://reviewable.io/reviews/d-ronin/dronin/595)
closed 2016-02-04 by tracernz